micropython/micropython #13167: Network - AttributeError: 'module' object has no attribute 'STAT_CONNECT_FAIL'

Jael-G, 2023-12-10:
According to the micropython library and docs, network has various status constants that are return whenever WLAN.status() is called. However, one of them should be STAT_CONNECT_FAIL. While mapping the different status constants, I get the error:

```
AttributeError: 'module' object has no attribute 'STAT_CONNECT_FAIL'
```

I tried searching if anyone had this issue as well, and didn't seem like it, and I doubt I'm doing anything wrong given that it's pretty straight forward. Here is this code:

```python
import network

status_names = {
    network.STAT_CONNECT_FAIL: 'STAT_CONNECT_FAIL',  
}
```

I'm guessing this is some sort of bug. I apologize in advance if somehow it's not a bug, it's a known issue, or I'm implementing it wrong.

Thanks.

Flipje1955, 2023-12-10:
stumbled into this, looking for the answer myself: https://github.com/micropython/micropython/issues/3006. 

Looking into network gives:  ['__class__', '__name__', 'AP_IF', 'STAT_CONNECTING', 'STAT_CONNECT_FAIL', 'STAT_GOT_IP', 'STAT_IDLE', 'STAT_NO_AP_FOUND', 'STAT_WRONG_PASSWORD', 'STA_IF', 'WLAN', '__dict__', 'country', 'hostname', 'route']....

jonnor, 2024-08-24:
Thank you for reporting the issue and how you resolved it - it makes it easier for others online to find the fix :)

thatbudakguy, 2024-09-28:
For anyone who runs this on MicroPython 1.24.0-preview on an ESP32, the closest existing error code seems like it might be `STAT_ASSOC_FAIL`? I don't have `STAT_CONNECT_FAIL`, but I was able to check what was there (as suggested above) with:
```python
>>> import network
>>> [member for member in dir(network) if member.startswith("STAT_")]
['STAT_ASSOC_FAIL', 'STAT_BEACON_TIMEOUT', 'STAT_CONNECTING', 'STAT_GOT_IP', 'STAT_HANDSHAKE_TIMEOUT', 'STAT_IDLE', 'STAT_NO_AP_FOUND', 'STAT_NO_AP_FOUND_IN_AUTHMODE_THRESHOLD', 'STAT_NO_AP_FOUND_IN_RSSI_THRESHOLD', 'STAT_NO_AP_FOUND_W_COMPATIBLE_SECURITY', 'STAT_WRONG_PASSWORD']
```

projectgus, 2024-10-23:
thanks @thatbudakguy. It seems like esp32 has `STAT_ASSOC_FAIL` and esp8266 and rp2 ports have `STAT_CONNECT_FAIL`. However the documentation doesn't include the esp32 constant (or some other constants added for ESP-IDF v5.2).

This is something we should fix in the `network.WLAN` docs (and consider adding `STAT_CONNECT_FAIL` to ESP32 for consistency.)

PR #16270 (2024-11-30): esp32: Add missing network.STAT_CONNECT_FAIL constant.
### Summary

The esp32 port had `network.STAT_ASSOC_FAIL` for the same purpose, but this is undocumented and different to all other ports which use the documented `STAT_CONNECT_FAIL`. This PR deprecates `STAT_ASSOC_FAIL` for removal in MicroPython 2.0.

Closes #13167

*This work was funded through GitHub Sponsors.*

### Testing

Flashed ESP32 port, checked that both constants exist and have the same numeric value.

### Trade-offs and Alternatives

Having two `STAT_` constants with the same numeric value might break someone's code if they'r [... cut]
closed 2024-11-30 by dpgeorge