micropython/micropython #13178: esp32s3 TouchPad not working

wangshujun-tj, 2023-12-12:
from machine import TouchPad, Pin
import time
t = TouchPad(Pin(5))
for i in range(100):
    time.sleep(0.5)
    print("%X"%t.read())

The same code changes the number with touch on ESP32, but the reading does not change on ESP32s3

Adjusting the initialization code as follows works fine, but I am not familiar enough with ESP32 and have not fully tested it, for reference only. Additionally, I suspect that the same issue also exists in ESP32s2
The changes are minimal, move the touch_Pad_Fsm_Start  to touch_Pad_config After 

STATIC mp_obj_t mtp_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw,
    const mp_obj_t *args) {
    mp_arg_check_num(n_args, n_kw, 1, 1, true);
    gpio_num_t pin_id = machine_pin_get_id(args[0]);
    const mtp_obj_t *self = NULL;
    for (int i = 0; i < MP_ARRAY_SIZE(touchpad_obj); i++) {
        if (pin_id == touchpad_obj[i].gpio_id) {
            self = &touchpad_obj[i];
            break;
        }
    }
    if (!self) {
        mp_raise_ValueError(MP_ERROR_TEXT("invalid pin for touchpad"));
    }

    static int initialized = 0;
    if (!initialized) {
        touch_pad_init();
        touch_pad_set_fsm_mode(TOUCH_FSM_MODE_TIMER);
        initialized = 1;
    }
    #if CONFIG_IDF_TARGET_ESP32
    esp_err_t err = touch_pad_config(self->touchpad_id, 0);
    #elif CONFIG_IDF_TARGET_ESP32S2 || CONFIG_IDF_TARGET_ESP32S3
    esp_err_t err = touch_pad_config(self->touchpad_id);
    touch_pad_fsm_start();
    #endif
    if (err == ESP_OK) {
        return MP_OBJ_FROM_PTR(self);
    }
    mp_raise_ValueError(MP_ERROR_TEXT("Touch pad error"));
}

wangshujun-tj, 2024-01-29:
Has any developer noticed this? I only tested esp32s3 and suspect that s2 also has the same issue, but I don't have the appropriate hardware at hand

x-channel, 2024-03-16:
I got same problem.

SkaveRat, 2024-04-06:
I *think* I got the same problem

When I read the value I get nonsensical values like "3252087".

Test code:

```
touch_pin = TouchPad(Pin(3, Pin.IN))
touch_pin.read()
```

wangshujun-tj, 2024-04-30:
我给出的帖子里面提到了解决方法，按照做简单修改既可以工作
可以得到可靠的可变的读数，但是s3和老的esp32数据变化趋势相反，数值范围差异也很大

Shahir-abdulla, 2024-05-16:
> from machine import TouchPad, Pin import time t = TouchPad(Pin(5)) for i in range(100): time.sleep(0.5) print("%X"%t.read())
>
> The same code changes the number with touch on ESP32, but the reading [...]

This actually works.

brainstorm, 2024-07-28:
I felt like I was chasing my tail with this one since it worked perfectly fine on a [WaveShare ESP32-S3-Zero board](https://www.waveshare.com/esp32-s3-zero.htm) with the following code:

```python
from machine import TouchPad, Pin
import time
t = TouchPad(Pin(1))

while True:
    time.sleep(0.5)
    print(t.read())
```

OTOH, on a custom board with a [ESP32-S3-MINI-1 module](https://www.espressif.com/sites/default/files/documentation/esp32-s3-mini-1_mini-1u_datasheet_en.pdf) I was seeing static values as other folks report in this issue such as the following, with no change at all when touching/not touching:

```
(...)
1957609
1957609
1957609
1957609
(..)
```

Then I connected a oscilloscope probe, and saw the following waveforms on the custom board:

## Baseline (no touch other than the scope probe):

[image]

## Touching with fingers:

[image]

And then I looked bacl at the waveforms of the waveshare board and saw this:

## No touch (only scope probe attached):

[image]

## Touch:

[image]

So I suspected that the difference on baselines between boards might be the culprit?

Then I thought: well, **perhaps folks initialize the touch too early before the baseline is relatively stable on custom boards (with more noise and design/manufacturing defects)? That'd also explain the relocation of the `touch_pad_fsm_start()` function and at least one user reporting success too.**

The following code (waiting for 10 seconds before importing `TouchPad` at all), the touch values make sense on the custom board:

```
42719
42759
178112
200197
218871
228830
235282
241806
238224
247305
247342
251422
42723
42731
```

PR #16169 (2024-11-28): esp32: Fix stuck TouchPad readings on ESP32-S3, use version specifiers in code.
### Summary

* Updates the `machine.TouchPad` driver to use hardware version specifiers instead of chip names. This should make adding future SoCs easier.
* Fixes #13178, by updating the fix applied in #8955.
* Update the docs for ESP32-S3 and ESP32-S3 TouchPad differences.

The new order of initialisation matches more closely the order in Espressif's example code, i.e.
https://github.com/espressif/esp-idf/blob/v5.2.2/examples/peripherals/touch_sensor/touch_sensor_v2/touch_pad_read/main/tp_read_main.c#L86

*This wo [... cut]