micropython/micropython #13182: VM stack overflow when importing frozen modules.

iabdalkader, 2023-12-12:
With `MICROPY_DEBUG_VM_STACK_OVERFLOW` enabled, importing frozen modules causes a VM stack overflow:

```
MicroPython v1.22.0-preview.255.g9e262d8f9.dirty on 2023-12-12; Arduino GIGA R1 WiFi with STM32H747
Type "help()" for more information.
>>> import time
VM stack overflow state=2407fc74 n_state+1=2
Assertion '0' failed, at file ../../py/objfun.c:302

FATAL ERROR:
```

I can't reproduce this with F7 only the H7 for some reason. Any idea what might be causing this or what I could do to debug this further ?

iabdalkader, 2023-12-12:
I was able to reproduce this with PYBD SF2, it doesn't happen with just any frozen module, it has to be a module that extends a built-in, so in PYBD manifest I added:
```python
require("time")
```
And now it overflows:

```
MicroPython v1.22.0-preview.255.g9e262d8f9.dirty on 2023-12-12; PYBD-SF2W with STM32F722IEK
Type "help()" for more information.
>>> import time
VM stack overflow state=2000ae44 n_state+1=2
```

@dpgeorge Is this VM debug feature just not updated, or is there an actual bug here ?

dpgeorge, 2023-12-12:
The `MICROPY_DEBUG_VM_STACK_OVERFLOW` option affects both the compiler/emitter and the runtime, so if bytecode is being generated by `mpy-cross` (ie frozen) then `mpy-cross` must also be compiled with this option enabled.  When this option is enabled it will reserve an extra slot in the Python stack frame of the function so it can detect stack overflow.

Did you enable it in the firmware and `mpy-cross` used to freeze the code?

iabdalkader, 2023-12-12:
> Did you enable it in the firmware and `mpy-cross` used to freeze the code?

Ah I see, I did now and it works, thank you!

closed 2023-12-12 by iabdalkader