micropython/micropython #13209: ESP32(S3) - littlefs - Consistant drop of module loading speed if files are written to filesystem

ftylitak, 2023-12-15:
Greetings all,

the past two days i am troubled by a weird behaviour pattern that I was not able to find any answer, so I am posting it here hopping to find a solution.

* __board__: ESP32-S3-WROOM-1 and ESP32-WROVER-IE
* __micropython__: 1.16, 1.19.1, 1.21
* __file system__: littlefs v2

I am running a very simple script.
The scenario is to wake up, write content to a file, load a module and deepsleep. 

My observation is that every time a file is written to the file system, the system speed is affected in a consistant pace. After each wakeup from deep sleep, running the same script takes more and more time to execute. At some point it starts executing the scirpt quickly again and starts gradually the speed slow down.

This is the main.py where at the end of the module import, before going to sleep just prints the ticks_ms to use as reference:

```py
import utime

destination = "/testing"
content = "00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000"

with open(destination, "w") as file:
    file.write(content)

from onewire import OneWire
from ds18x20 import DS18X20
print("load time: {}".format(utime.ticks_ms()))
utime.sleep_ms(1000)
import machine
machine.deepsleep(1)
```

And here is the behaviour in multiple setups shown in graphs, where the Y axis is the execution time printed by the line `print("load time: {}".format(utime.ticks_ms()))`  and the X axis is the iteration number:

* ESP32-S3-WROOM-1 (micropython 1.21.0) - littlefs v2
   * [image]
   
* ESP32-S3-WROOM-1 (micropython 1.19.1) - littlefs v2
   * [image]

* ESP32-WROVER-IE (micropython 1.16) - littlefs v2
   * [image]

* ESP32-S3-WROOM-1 (micropython 1.19) - fat
   *  [image]

__Notes__
* This has to do with the littlefs somehow. fat does not have this behaviour
* The loading time is directly dependand to some state of the file system because even after a power off/on, the loading speed follows exactly the pattern from where it was before power off. 
* Different content length when writting, affects the rythm of speed decline

An help will be very much appreciated as I have hit a deadend to my investigation.

andrewleech, 2023-12-15:
To a large extent this is probably the standard behaviour of littles. The safety / stability of it does come as cost: https://github.com/littlefs-project/littlefs/blob/master/DESIGN.md

robert-hh, 2023-12-16:
@ftylitak See as well the discussion in #13088, which made a slightly different test for FAT file writes of many files and similar behavior when using LFS.

peterhinch, 2023-12-17:
Apologies in advance if I'm missing something, but it's not clear to me what
```py
print("load time: {}".format(utime.ticks_ms()))
```
is actually measuring. To measure time intervals you normally use `ticks_diff` as follows:
```py
from time import ticks_diff, ticks_ms
start = ticks_ms()
# do something
print(f"Duration {ticks_diff(ticks_ms(), start)}ms")
```
In other words you measure an elapsed time rather than an absolute time.

ftylitak, 2023-12-17:
@peterhinch, to my understanding `ticks_ms` provides the milliseconds since the boot of the os. In my example, the device enters in deepsleep, upon wakeup it will load the module and print the elapsed time since the os boot. Isn't it?

@andrewleech & @peterhinch thank you for your references. It seems it is a complex issue and it will take me time to go through the details of littlefs oepration. 

Till then i am thinking of creating a separate partition to split the code from the stored measurements/flag files hopping to avoid affecting the code execution speed just because of the stored data. I will keep you updated.

robert-hh, 2023-12-17:
> o my understanding ticks_ms provides the milliseconds since the boot of the os. In my example, the device enters in deepsleep, upon wakeup it will load the module and print the elapsed time since the os boot. Isn't it?

Peter#s point here might be is, that you do not measure the time of the individual steps, of:
- (hidden) mounting of the file system. For that the first ticks_ms() could be used, which indeed starts at 0 at every hard reset, deepsleep() or machine.reset().
- Writing the file
- loading the modules.

As a guess I would assume, that the repeating file write is the major cause for the timing pattern.

ftylitak, 2023-12-18:
@robert-hh & @peterhinch ok I get your point. The example code that I posted above can be a bit missleading. 

For the sake of completeness let me update the code with the correct timing of the behaviour that I want to address.

```py
from utime import ticks_diff, ticks_ms, sleep_ms

start = ticks_ms()
from onewire import OneWire
from ds18x20 import DS18X20
print(f"Module load duration {ticks_diff(ticks_ms(), start)}ms")

destination = "/stored_messurements"
content = '{"temp":"12.25","t":1702890685}'

with open(destination, "w") as file:
    file.write(content)

sleep_ms(1000)
import machine
machine.deepsleep(1)
```
and here is the loading times of this this example for esp32s3 and micropython 1.19.1 on littlefs:

[image]

Indeed it is clear that the file writes cause the behavior. My actual scenario is that the device executes a pack of measurements, stores them locally and enters deepsleep for 1 hour. Every 24 hours it will upload the stored measurements. Each pack of measurements reports the time the device stays awake before it gets to deepsleep (uptime) and here is an example of the  reported values:

[image]

My conclusion is that I need to separate the code files from the generated files. Thus there will be a partition for the source files that will be written only in case of OTA in the hope that the source code will have consistent loading times across runs.

ftylitak, 2024-01-22:
I wanted to share an update of the workaround that worked for us.

We finally ended up with the solution of using separate partition for the generated measurement storage, to avoid affecting the loading speed of the code.

First step, add a new data partition in the partition table of the ESP32(S3). The following example corresponds to 8MiB flash:
```
# Notes: the offset of the partition table itself is set in
# $IDF_PATH/components/partition_table/Kconfig.projbuild.
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 0x26FF00,
vfs,      data, fat,     0x280000, 0x31FF00,
data,     data, fat,     0x5A0000, 0x10FFFF, # the new partition
```

Second step, on boot, mount the partition if found under /data folder:

```py
# setup data paritition
from esp32 import Partition
p = Partition.find(Partition.TYPE_DATA, label='data')[0]

try:
    if p:
        uos.mount(p, "/data")
        print("[boot] Mounted /data")
    else:
        print("[boot] Data partition not found")
except OSError:
    try:
        uos.VfsLfs2.mkfs(p)
        vfs = uos.VfsLfs2(p)
        uos.mount(vfs, '/data')
        print("[boot] Created + Mounted /data")
    except OSError:
        print("[boot] Failed mounting /data")
```

Now, when storing under /data, the code execution time (ex. module loading time) is not affected.