micropython/micropython #13210: RP2: Floats smaller than 1e-37 printed wrongly (and subnormals treated as 0.0)

rkompass, 2023-12-16:
In completing my function for showing the representation of floats for the case of subnormals:
```py
def fp32_repr(x, bits=True):
    import struct
    a0 = int.from_bytes(struct.pack('f', x), 'little')
    if bits:
        return f'{(a0>>31)&1:01b} {((a0>>23) & 0xff):08b} {(a0 & 0x7fffff):023b}'
    else:
        sign = '-' if a0 >> 31 else ''
        exp = ((a0>>23) & 0xff)
        m = a0 & 0x7fffff
        if 255 > exp > 0:
            exp -= 150
            m |= 0x800000
        elif exp == 0:   # subnormal
            exp -= 149
        elif exp == 255: # Inf or NaN
            if m == 0:
                return f'{sign}Inf'
            else:
                return 'NaN'
        return f'{sign}{m} * 2**{exp}'
```
I noted that the RP2040 port (as of v1.22.0-preview.226.g0960d64d3 on 2023-12-06), after correction of the last float imprecision issue handles small floats differently than the other ports in that it treats subnormal representations as zero (0.0). This might be documented as a feature, not a bug:-), if correction seems too difficult.

A real bug is the printing of very small floats (< 1e-37, but still normal representation i.e. exponent bits > 0) as 1e-37.
This is demonstrated by the following code:

```py
def irep2fp32(i):
    import struct
    return struct.unpack('f',i.to_bytes(4, 'little'))[0]

# With correct implementation of fp32: (including subnormal representation, where exp bits are 0)
# e.g. on Pyb 1.1 / Blackpill (stm32f411):
#
# int.from_bytes(struct.pack('f', 4.2e-41),'little')  -->    29972  subnormal
# int.from_bytes(struct.pack('f', 4.2e-39),'little')  -->  2997220  subnormal
# int.from_bytes(struct.pack('f', 3.2e-38),'little')  --> 19806589     normal
# int.from_bytes(struct.pack('f', 1.2e-37),'little')  --> 35870182     normal

print(irep2fp32(29972), irep2fp32(29972)*1000)
print(irep2fp32(2997220), irep2fp32(2997220)*1000)
print(irep2fp32(19806589), irep2fp32(19806589)*1000)
print(irep2fp32(35870182), irep2fp32(35870182)*1000)

print(3.2e-38)
```
On Blackpill and w600  boards (correct):
```
4.200112e-41 4.199972e-38
4.199999e-39 4.2e-36
3.2e-38 3.2e-35
1.2e-37 1.2e-34
3.2e-38
```
on esp8266 (30 bit):
```
4.20011e-41 4.19997e-38
4.2e-39 4.2e-36
3.2e-38 3.2e-35
1.2e-37 1.2e-34
3.2e-38
```

on Raspberry Pi Pico with RP2040 MicroPython v1.22.0-preview.226.g0960d64d3 on 2023-12-06
```
1e-37 0.0         <-- subnormal (4.2e-41), interpreted as 0.0 in calculation, but printed as 1e-37
1e-37 0.0         <-- subnormal (4.2e-39), interpreted as 0.0 in calculation, but printed as 1e-37
1e-37 3.2e-35     <-- first is incorrect, although not subnormal, probably only wrong print
1.2e-37 1.2e-34   <-- correct
1e-37                    <-- wrong, bug
```

dpgeorge, 2023-12-17:
Thanks for the report.  It looks like the pico-sdk doesn't handle subnormals at all, which seems strange...

Here's a slightly simpler test showing the same issue:
```py
import struct

def fhex(f):
    return struct.pack("f", f).hex()

a = struct.unpack("f", bytes.fromhex("14750000"))[0]
b = struct.unpack("f", bytes.fromhex("e4bb2d00"))[0]

print(a, fhex(a), fhex(a + 0), fhex(a * 1))
print(b, fhex(b), fhex(b + 0), fhex(b * 1))
```

Correct output (PYBv1.0 with hardware single precision float):
```
4.200112e-41 14750000 14750000 14750000
4.199999e-39 e4bb2d00 e4bb2d00 e4bb2d00
```
Output on a Pico:
```
1e-37 14750000 00000000 00000000
1e-37 e4bb2d00 00000000 00000000
```
The printing of the subnormal is wrong, but I think that's just because the pico-sdk doesn't handle subnormals, as you can see above, adding 0 or multiplying by 1 truncates the subnormal to 0.

dpgeorge, 2023-12-17:
Bug reported to pico-sdk: https://github.com/raspberrypi/pico-sdk/issues/1582

To work around this bug we would need to bypass the RP2040 float ROM routines for all cases where subnormal values are input in to them.