micropython/micropython #13220: heap-buffer-overflow [micropython@a5bdd39127]

zeroone-kr, 2023-12-18:
Hi all, I'm Wonil Jang, from the research group S2Lab in UNIST.
We found a heap buffer overflow bug from micropython by our custom tool. The detailed information is as follows.

## Environment
- OS: Ubuntu 22.04
- Version: micropython at commit a5bdd39127
- Build: unix ports
- Bug Type: heap-buffer-overflow
- Bug Location: [micropython/lib/re1.5/compilecode.c:68](https://github.com/micropython/micropython/blob/master/lib/re1.5/compilecode.c#L68)
- Detection Tool: our custom tool, not disclosed yet
- Credits: Junwha Hong and Wonil Jang, from S2Lab in UNIST

## Problem Statement
```cpp
...
case '[': {
            int cnt;
            term = PC;
            re++;
            if (*re == '^') {
                EMIT(PC++, ClassNot);
                re++;
            } else {
                EMIT(PC++, Class);
            }
            PC++; // Skip # of pair byte
            prog->len++;
            for (cnt = 0; *re != ']'; re++, cnt++) {    <== [1]
                char c = *re;
                if (c == '\\') {
                    ++re;
                    c = *re;
                    if (MATCH_NAMED_CLASS_CHAR(c)) {
                        c = RE15_CLASS_NAMED_CLASS_INDICATOR;
                        goto emit_char_pair;
                    }
                }
                if (!c) return NULL;
                if (re[1] == '-' && re[2] != ']') {    <== [2]
                    re += 2;
                }
            emit_char_pair:
                EMIT(PC++, c);
                EMIT(PC++, *re);
            }
            EMIT_CHECKED(term + 1, cnt);
            break;
        }
...
```
At [1], We found a heap-buffer-overflow bug. This is because there is no verification of regex string length. 
For instance, If we have a regex string ```([a-zA-Z_0-(null)```, it will process ```0-(null)``` finally.
and then, at [2], ```re``` pointer will point to next byte after ```0-(null)```.
Because It doesn’t have verification logic of regex string length, it will try to read a byte after ```0-(null)```, which means out-of-bound read.

## Example (replication)

```python
>>> import re
>>> re.sub(r"([a-zA-Z_0-", "b", "1a2a3a")
=================================================================
    #0 0x5555557b9461 in _compilecode /home/qbit/testing-2023/micropython/ports/unix/../../lib/re1.5/compilecode.c:68:27
    #1 0x5555557b97ed in _compilecode /home/qbit/testing-2023/micropython/ports/unix/../../lib/re1.5/compilecode.c:103:18
    #2 0x5555557b78da in re1_5_sizecode /home/qbit/testing-2023/micropython/ports/unix/../../lib/re1.5/compilecode.c:194:9
    #3 0x5555557b78da in mod_re_compile /home/qbit/testing-2023/micropython/ports/unix/../../extmod/modre.c:427:16
    #4 0x5555557b6cd7 in re_sub_helper /home/qbit/testing-2023/micropython/ports/unix/../../extmod/modre.c:287:16

```

## Patch
The simplest patch will be adding verification of regex ```re``` length.

## Log
```
    #0 0x5555557b9461 in _compilecode /home/qbit/testing-2023/micropython/ports/unix/../../lib/re1.5/compilecode.c:68:27
    #1 0x5555557b97ed in _compilecode /home/qbit/testing-2023/micropython/ports/unix/../../lib/re1.5/compilecode.c:103:18
    #2 0x5555557b78da in re1_5_sizecode /home/qbit/testing-2023/micropython/ports/unix/../../lib/re1.5/compilecode.c:194:9
    #3 0x5555557b78da in mod_re_compile /home/qbit/testing-2023/micropython/ports/unix/../../extmod/modre.c:427:16
    #4 0x5555557b6cd7 in re_sub_helper /home/qbit/testing-2023/micropython/ports/unix/../../extmod/modre.c:287:16
    #5 0x555555782c1c in mp_execute_bytecode /home/qbit/testing-2023/micropython/ports/unix/../../py/vm.c:1042:21
    #6 0x55555574261b in fun_bc_call /home/qbit/testing-2023/micropython/ports/unix/../../py/objfun.c:273:42
    #7 0x555555903f3d in execute_from_lexer /home/qbit/testing-2023/micropython/ports/unix/main.c:161:13
    #8 0x555555902ad5 in do_file /home/qbit/testing-2023/micropython/ports/unix/main.c:310:12
    #9 0x555555902ad5 in main_ /home/qbit/testing-2023/micropython/ports/unix/main.c:722:19
    #10 0x7ffff7c29d8f in __libc_start_call_main csu/../sysdeps/nptl/libc_start_call_main.h:58:16
    #11 0x7ffff7c29e3f in __libc_start_main csu/../csu/libc-start.c:392:3
    #12 0x555555593a34 in _start (/home/qbit/testing-2023/micropython/ports/unix/build-standard/micropython+0x3fa34)
```

projectgus, 2024-01-31:
Hi @zeroone-kr .

This is accounted for in the check at compilecode.c:79:

> if (!c) return NULL;

Because of this check, the next line (marked `[2]` in your submission) is never executed for a value of `re` that points after the terminating 0 byte.

Similarly, if I execute your `Example (replication)` code then it prints `ValueError: error in regex` as expected.

zeroone-kr, 2024-01-31:
Hi @projectgus, 
Thank you for your kind response and effort on this issue.
By the way, this issue is constructed with turning on GIL in unix because we also noticed that the bugs from race conditions from absence of GIL would be not a vulnerability.

Thus, above bug can be reproduced with GIL setting on Linux, with following commands.
```bash
sed -i 's/CFLAGS += -DMICROPY_PY_THREAD=1 -DMICROPY_PY_THREAD_GIL=0/CFLAGS += -DMICROPY_PY_THREAD=1 -DMICROPY_PY_THREAD_GIL=1/' ports/unix/Makefile
sed -i 's/#define MICROPY_ASYNC_KBD_INTR         (1)/#define MICROPY_ASYNC_KBD_INTR         (0)/' ports/unix/variants/mpconfigvariant_common.h
```
And as you said, we also checked the `ValueError: error in regex` is printed,
even though It seems like there is no problem, if we debug the program, it is observed that `re` points to invalid address.
To describe in detail, if the pattern `([a-zA-Z_0-` is provided, the values in variables cnt, re, and c are changing as follows at [compilecode.c:70](https://github.com/micropython/micropython/blob/master/lib/re1.5/compilecode.c#L70). (analyzed in gdb)

```cpp
for (cnt = 0; *re != ']'; re++, cnt++) {    // <== compilecode.c:68
    char c = *re;
    if (c == '\\') {    // <== compilecode.c:70
        ++re;
        c = *re;
        if (MATCH_NAMED_CLASS_CHAR(c)) {
            c = RE15_CLASS_NAMED_CLASS_INDICATOR;
            goto emit_char_pair;
        }
    }
    if (!c) return NULL;  // <== compilecode.c:78
    if (re[1] == '-' && re[2] != ']') {  // <== compilecode.c:79
        re += 2;    // <== compilecode.c:80
    }    
emit_char_pair:
    EMIT(PC++, c);
    EMIT(PC++, *re);
}
```

|cnt|(re, c) state at compilecode.c:70|
|----|----|
|0|("a-zA-Z_0-(null)", 'a')|
|1|("A-Z_0-(null)", 'A')|
|2|("\_0-(null)", '_')|
|3|("0-(null)", '0')|

The problem is that, at the iteration 3(cnt=3), the `re` will point to "(null)" by `re += 2` statement at [compilecode.c:80](https://github.com/micropython/micropython/blob/master/lib/re1.5/compilecode.c#L80) because there is no validation about nullstring at compilecode.c:79 (re[1] == '-' && re[2] != ']') and just jump to next next position. 
That's why the validation on !c at line 78, which you mentioned, is not applied for this PoC

Now, the invalid memory address `re` is accessed after increasing `re` pointer by 1 at `for (cnt = 0; *re != ']'; re++, cnt++){ `, which is located at  [compilecode.c:68](https://github.com/micropython/micropython/blob/master/lib/re1.5/compilecode.c#L68).

For your convenience to reproduce this issue, we prepared a gdb script in below. (reproduce_poc_by_gdb)
```
define pv
c
p cnt
p re
p c
end

b compilecode.c:70
r
set $pattern=re
pv
pv
pv

b compilecode.c:68
c
p $pattern
p re
```
and the poc.py is same as before (poc.py)
```python
import re
re.sub(r"([a-zA-Z_0-", "b", "1a2a3a")
```
you can run the gdb script by `gdb -x reproduce_poc_by_gdb--args [your_micropython_executable] poc.py`

Would you have the time to review this issue once more at your convenience?