micropython/micropython #13238: Zephyr port is not compatible with Zephyr v3.2.x or higher (latest v3.5.0).

GunterO, 2023-12-20:
When using Zephyr 3.5.0 (latest), I get an error when trying to build this:
`west build -b qemu_x86 ~/micropython/ports/zephyr --pristine=always`

Error:
`FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/go/zephyrproject/zephyr/build`

Complete build log:
[build_error_v3.5.0.txt](https://github.com/micropython/micropython/files/13726107/build_error_v3.5.0.txt)

I have tried the above on Windows 10 Pro & a clean install of Ubuntu 22.04 LTS, same result.
Tried with v3.2.0 and v3.4.0 too, same problem. Only v3.1.0 seems to work, but is missing the m5stack_core2 target which I need.

When I checkout Zephyr v3.1.0, the same build is completed without any error.
```
git checkout v3.1.0
west update
west build -b qemu_x86 ~/micropython/ports/zephyr --pristine=always
```
Anybody an idea to fix this? Thanks!

GunterO, 2023-12-20:
When I run the build on Zephyr 3.5.0 a second time, I get another error...
Same when I try to build it a third, forth, ... time. Probably this is the real error?
```
[243/466] Generating genhdr/qstr.i.last
FAILED: genhdr/qstr.i.last /home/go/zephyrproject/zephyr/build/genhdr/qstr.i.last 

```
Complete build log:
[build_error_v3.5.0_b.txt](https://github.com/micropython/micropython/files/13726249/build_error_v3.5.0_b.txt)

jimmo, 2023-12-21:
This is a known issue. Only Zephyr 3.1 is currently supported.

I have it working with latest Zephyr in a branch but was trying to fix a few other outstanding issues with the Zephyr port and haven't had time to get it all done yet. Soon!

closed 2023-12-21 by jimmo

MaureenHelm, 2024-01-03:
See #9335