micropython/micropython #13249: rp2 PIO get and put non-blocking

nludban, 2023-12-21:
My application wants to do non-blocking i/o between `array` and the PIO state machines.  The rp2 port only supports blocking and full length i/o, so I have to check the FIFO levels before doing a `get` or `put` to avoid blocking, resulting in duplicated code and activity.

Basic request is to support a timeout for the operation and return a short count - maybe from a new method to avoid changing return types.

Would prefer `get` to support `MIN` and `TIME` semantics defined by [termios(3)](https://man7.org/linux/man-pages/man3/termios.3.html) (but with standard time units).

As an extra convenience, optionally wait for the txfifo to be emptied within the timeout.

mendenm, 2023-12-24:
You can effectively do this with fifo():
```python
        while 1:
            nfifo = int(fifo())
            if nfifo != 0: break
```
and put (if you want) an asyncio sleep inside the loop.  Then read the number of bytes in nfifo.
If you do this in viper mode, it's pretty fast.

jonnor, 2025-03-03:
There seems to be a rather acceptable solution provided above?

nludban, 2025-03-04:
> There seems to be a rather acceptable solution provided above?

I'm getting a burst of a few tens of words, 10 microseconds per word, so application level polling would be a significant overhead relative to the PIO FIFO capacity.

The bursts occur at 20 or 50 Hz, once all the data is received it must return to processing with minimal latency.

No data or a short read is a severe fault, it needs to detect these and handle the error within a few tenths of a second.

mendenm, 2025-03-04:
Can you just set up a DMA channel to capture the input, and then poll the DMA transfer counter to see when a packet is in? You  can set the DMA transfer count bigger than the biggest packet, and then restart it after you have captured a packet.  This would give hard real-time  performance with no processor load handling the transfer.