micropython/micropython #13260: REPL works on serial but stucks on thonny from bare-arm port

lwshowl, 2023-12-22:
[image]

thonny prints the welcome message but no input prompt is printed , and always shows the device is busy

[image]
in putty everything is fine , and tested working with a lot of functions

i ported from bare-arm ports and run the REPL loop only in main methods where GC are not enabled
because i got compile errors using gc_helpers_generic.c , i seems not to compile arm code

code
```c

int main(int argc, char **argv) {
    // Initialise the MicroPython runtime.
    mp_stack_ctrl_init();
    mp_init();

    // Start a normal REPL; will exit when ctrl-D is entered on a blank line.
    pyexec_friendly_repl();

    mp_deinit();
    return 0;
}

// Handle uncaught exceptions (should never be reached in a correct C implementation).
void nlr_jump_fail(void *val) {
    for (;;) {
    }
}

// There is no filesystem so stat'ing returns nothing.
mp_import_stat_t mp_import_stat(const char *path) {
    return MP_IMPORT_STAT_NO_EXIST;
}

// There is no filesystem so opening a file raises an exception.
mp_lexer_t *mp_lexer_new_from_file(const char* filename) {
    mp_raise_OSError(MP_ENOENT);
}
```

dpgeorge, 2023-12-23:
The bare-arm port is not intended to be a fully-functioning port with all the features enabled.  It probably won't work with Thonny for that reason.

But this is anyway an issue with Thonny, not with MicroPython.

closed 2023-12-23 by dpgeorge

lwshowl, 2023-12-24:
> The bare-arm port is not intended to be a fully-functioning port with all the features enabled. It probably won't work with Thonny for that reason.
> 
> But this is anyway an issue with Thonny, not with MicroPython.

thanks , what are the features should i implement to have it work with thonny ?

andrewleech, 2023-12-24:
> > The bare-arm port is not intended to be a fully-functioning port with all the features enabled. It probably won't work with Thonny for that reason.
> > 
> > But this is anyway an issue with Thonny, not with MicroPython.
> 
> thanks , what are the features should i implement to have it work with thonny ?

You would need to ask the Thonny project that question, only they know what features they need.

lwshowl, 2023-12-24:
> You would need to ask the Thonny project that question, only they know what features they need.

ok, thanks

robert-hh, 2023-12-24:
That's what Thonny is doin on device startup (ESP32). Numbers in [] are non-printable characters, like [1] for Ctrl-A and [3] for Ctrl-C, ..... That is all quite a lot. But it tells what Thonny expects from a device.:
```
[03][01][02][01][05]A[01]class __thonny_helper:
    import builtins
    try:
        import uos as os
    except builtins.ImportError:
        import os
    import sys
    last_non_none_repl_value = None

    # for object inspector
    inspector_values = builtins.dict()
    @builtins.classmethod
    def print_repl_value(cls, obj):
        if obj is not None:
            cls.builtins.print('[object_link_for_thonny=%d]' % cls.builtins.id(obj), cls.builtins.repr(obj), '[/object_link_for_thonny]', sep='')
            cls.last_non_none_repl_value = obj

    @builtins.classmethod
    def print_mgmt_value(cls, obj):
        cls.builtins.print('<thonny>', cls.builtins.repr(obj), '</thonny>', sep='', end='')

    @builtins.classmethod
    def repr(cls, obj):
        try:
            s = cls.builtins.repr(obj)
            if cls.builtins.len(s) > 50:
                s = s[:50] + "..."
            return s
        except cls.builtins.Exception as e:
            return "<could not serialize: " + __thonny_helper.builtins.str(e) + ">"

    @builtins.classmethod
    def listdir(cls, x):
        if cls.builtins.hasattr(cls.os, "listdir"):
            return cls.os.listdir(x)
        else:
            return [rec[0] for rec in cls.os.ilistdir(x) if rec[0] not in ('.', '..')]

    @builtins.classmethod
    def getcwd(cls):
        return cls.os.getcwd()

    @builtins.classmethod
    def chdir(cls, x):
        return cls.os.chdir(x)

    @builtins.classmethod
    def rmdir(cls, x):
        return cls.os.rmdir(x)
[04][05]A[01]__thonny_helper.print_mgmt_value(__thonny_helper.getcwd())[04][05]A[01]__thonny_helper.print_mgmt_value(__thonny_helper.sys.path)[04][05]A[01]__thonny_helper.builtins.help('modules')[04][05]A[01]
try:
    from time import localtime as __thonny_localtime
    __thonny_helper.print_mgmt_value(__thonny_helper.builtins.tuple(__thonny_localtime(946684800)))
    del __thonny_localtime
except __thonny_helper.builtins.Exception as e:
    __thonny_helper.print_mgmt_value(__thonny_helper.builtins.str(e))
[04][05]A[01]
try:

    from machine import RTC as __thonny_RTC
    try:
        __thonny_RTC().datetime((2023, 12, 24, 6, 16, 25, 15, 0))
    except:
        __thonny_RTC().init((2023, 12, 24, 16, 25, 15, 0, 0))
    finally:
        del __thonny_RTC

    __thonny_helper.print_mgmt_value(True)
except __thonny_helper.builtins.Exception as e:
    __thonny_helper.print_mgmt_value(__thonny_helper.builtins.str(e))
[04][05]A[01]__thonny_helper.print_mgmt_value({name : (__thonny_helper.repr(value), __thonny_helper.builtins.id(value)) for (name, value) in __thonny_helper.builtins.globals().items() if not name.startswith('__')})[04]
```

And that is send at every Enter in REPL:
```
[05]A[01]
[04][05]A[01]__thonny_helper.builtins.print('__thonny_helper' in __thonny_helper.builtins.dir())[04][05]A[01]__thonny_helper.print_mgmt_value({name : (__thonny_helper.repr(value), __thonny_helper.builtins.id(value)) for (name, value) in __thonny_helper.builtins.globals().items() if not name.startswith('__')})[04]
```