micropython/micropython #13268: SD read very slow on RP2040 w/sdcard.py driver

RaspberryPiDude314, 2023-12-24:
Obligatory Micropython/programming noob here. I'm working on a project involving playing video back on a Pi Pico running Micropython. For purposes of initial testing, I'm using https://github.com/russhughes/st7789_mpy for frame playback in JPG form on a ST7735 160*80 display. To make video playback easy, the workflow will be to first take a video or series of JPG images, decompressed shortly before playback when idle, and save these frames to an SD card. The original compressed images would also be stored in a separate directory on this SD card. During playback every decompressed image would be sequentially pulled from the SD card and displayed, then deleted/removed from RAM. This would mean decompression would not be a bottleneck while playing, as my tests found decompressing to be somewhere around 5 hertz, which is slower than I want for this project. 
The issue I'm having is that the sdcard.py driver is simply too slow to transfer these decompressed files from the card to RAM. Right now, the frames are written directly to individual files as bytearrays using the "wb" argument, which is plenty fast enough for me as a preparatory step. Each takes up around ~25KB of space. However, reading these back is incredibly slow in my tests. The fastest I've accomplished is around 0.332 seconds, ~3FPS, which is not nearly fast enough. I've tried every block size (32KB appears to be fastest by a few ms) and several reputable SD cards of different sizes. I've also tried bumping both machine.freq() and the SPI speed to their limits, the latter of which had no effect at all.
[A benchmark here from 2016](https://github.com/micropython/micropython/issues/2648) used the C SDCard library which isn't available for Pi Pico (not sure why) and was incredibly fast. The same issue also mentions sdcard.py but does not test those speeds.
If you have a suggestion for something I can do to speed this up, please let me know- I can't progress much in this project without fixing this problem. 
Thanks in advance!
I've also attached my (terrible) testing code below for reference.
`def fpstest():
    while True:
        t1=time.ticks_ms()
        with open("/sd/test02.txt", "rb") as file:
            data=file.read()
        t2=time.ticks_ms()
        print((t2-t1)/1000)`

RaspberryPiDude314, 2023-12-24:
Update: I realized that sdcard.py's init takes a baudrate, facepalm. However, I can only roughly double the speeds I'm getting to a max of about 0.137 seconds per read before it starts spitting out ENOENTs, the highest baudrate that worked reliably was about 20000000 (where I get around 0.15 seconds). That's not particularly slow for SPI but still much slower than RP2040's claimed maximum SPI baudrate. In addition, it certainly doesn't scale linearly, it's only twice as fast as the default baudrate of 1320000 while being 15 times the baudrate. I have no clue what's causing this behavior. SoftSPI was as expected much slower at the same set baudrate, more than half a second.

mendenm, 2023-12-25:
The rp2040 SPI doesn't operate at that speed.  I think the fastest it can go is 24 MHz or so, using the hardware SPI.  That in theory limits yuo to about 3 MB/s (ignoring overhead), but there is quite a bit of setup time on a transfer, and the transfers are only 512B, so the setup is quite expensive relative to the actual data move,  There is less setup time on multi-block transfers, so using big buffers helps some.

You may, though, want to try my updated SD driver, which is submitted as a pull request in:
https://github.com/micropython/micropython-lib/pull/765
It will run a little bit faster.  I can get about 900 kB/s transfer on reads, because it has somewhat less overhead in setting up the transfers. It may also be compatible with some cards that the current driver doesn't handle right.

jonnor, 2025-03-03:
Are you able to test the suggested MR above?