micropython/micropython #13271: 根据ESP32内存 编译多个MicroPython固件版本

BlackCattleProgrammer996, 2023-12-25:
MicroPython的工程师 圣诞节快乐! 
很荣幸有机会能够向你们反馈我的一些想法, 
ESP32有许多个 内存规格, 但是烧录某个固件 可能无法发挥出ESP32的内存; 我们现在的方法是修改并编译MicroPython源码 , 生成出适合 大容量的ESP32 , 希望在未来能够有多个不同规格的MicroPython固件 来适应不同的ESP32

bjsiyong, 2023-12-26:
老哥，搭车问一下，我现在在windows 11下编译 micropython。使用wsl2的unbuntu 20.04.6 LTS版本，esp-idf版本为release-5.2。
最后make的时候，提示：micropython/ports/esp32/network_common.c:172:1: error: static assertion failed: "Synchronize WIFI_AUTH_XXX constants with the ESP-IDF. Look at esp-idf/components/esp_wifi/include/esp_wifi_types.h" 
这个问题怎么解决呀？

bjsiyong, 2023-12-26:
> 老哥，搭车问一下，我现在在windows 11下编译 micropython。使用wsl2的unbuntu 20.04.6 LTS版本，esp-idf版本为release-5.2。 最后make的时候，提示：micropython/ports/esp32/network_common.c:172:1: error: static assertion failed: "Synchronize [...]

搞定了，在esp-idf/components/esp_wifi/include/esp_wifi_types.h代码中的WIFI_AUTH_MAX后面加上“=11”即可。如果你的micropython是v1.21.0则改为“=11”。其实删除也可以！

robert-hh, 2023-12-26:
See as well pull request #13197.

closed 2024-01-26 by BlackCattleProgrammer996