micropython/micropython #13273: ImportError: can't import name PWM

LinSakuraa, 2023-12-25:
```
PS D:\MicroPython\F407VE\Projects\demo\PWM_Demo> cli.exe -p COM14 repl

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: can't import name PWM
>>> 
```
I am trying to use the STM32F407VET6 to modify a pwm controller, but meet this error message when running the code. firmware is base STM32F4DISC and i add ```#define MICROPY_HW_ENABLE_PWM       (1) ```in ```mpconfigboard.h``` refrence https://github.com/micropython/micropython/issues/8059
Is this because the port of STM32F4 is not support PWM or I use it wrongly? plz tell me, thanks.

andrewleech, 2023-12-25:
PWM works differently on the stm port, `machine.PWM` doesn't yet exist 

Take a look at: https://github.com/orgs/micropython/discussions/11586

LinSakuraa, 2023-12-26:
Thanks,it works

closed 2023-12-26 by LinSakuraa