micropython/micropython #13280: Regression: I2C readfrom_mem throws EPERM on Portenta C33

sebromero, 2023-12-27:
I'm on `MicroPython v1.22.0 on 2023-12-27; Arduino Portenta C33 with RA6M5` and I just ran
a simple I2C read that worked fine under 1.21. It now throws an exception `EPERM`. It does however still work with SoftI2C.
I wonder what changed in the mean time in the Renesas port 🤔 
To reproduce it, run the following with the values adjusted for you I2C peripheral:

```
from machine import I2C, Pin, SoftI2C

buses = [I2C(0), SoftI2C(scl=Pin('P408') , sda=Pin('P407'))]

for bus in buses:
    print("Trying bus", bus)
    try:
        data = bus.readfrom_mem(0x21, 0x1c, 0x1) 
        print(data)
        print("Success on bus", bus)
    except Exception as e:
        print(f"Read failed on bus {bus}: {e}")
```

With my setup it prints the following:

```
Trying bus I2C(0, freq=400000, scl=P408, sda=P407)
Read failed on bus I2C(0, freq=400000, scl=P408, sda=P407): [Errno 1] EPERM
Trying bus SoftI2C(scl=P408, sda=P407, freq=500000)
b'\xb8'
Success on bus SoftI2C(scl=P408, sda=P407, freq=500000)
```

sebromero, 2023-12-27:
@iabdalkader Can you confirm this issue?

iabdalkader, 2023-12-27:
> @iabdalkader Can you confirm this issue?

I don't have any I2C sensor that I can easily use for testing. Do you have pull-ups on the device ? The board doesn't have any on that bus.

dpgeorge, 2023-12-27:
@sebromero can you run a git-bisect between 6117aa686f5faac6692f0def91be8e40a696aca3 (assuming that commit works) and master, and run your above script?  If there is a commit that introduced the regression, git-bisect will find it.

The only thing that touched I2C on renesas-ra was 7b2f13fb69d95980c8e517d1944bc1e9d7ed5489, but I can't see how that would break hardware I2C.

And as @iabdalkader mentions, make sure you have pull-up resistors on both SCL and SDA.

sebromero, 2024-01-08:
My bad, it's not a regression, I didn't test carefully enough. But it's still an issue. 
The pull up resistors are mounted on the client device (10k).
[image]

I tested once again and there is some weird behaviour that I'm not sure how to interpret. After a power cycle an I2C scan over hardware I2C works. But if I then call `.readfrom_mem(33, 13, 1)` it fails. A subsequent scan than keeps failing.
However, a software I2C scan and read still works after the hardware one fails. See example below: 

```
>>> from machine import I2C, Pin, SoftI2C
>>> I2C(0).scan()
[33]
>>> I2C(0).readfrom_mem(33, 13, 1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 1] EPERM
>>> I2C(0).scan()
[]
>>> SoftI2C(scl=Pin('P408') , sda=Pin('P407')).scan()
[33]
>>> SoftI2C(scl=Pin('P408') , sda=Pin('P407')).readfrom_mem(33, 13, 1)
b'\x01'
>>> 
```

Not sure what could be the problem. It does work with a regular Arduino sketch.

iabdalkader, 2024-01-09:
I think this is a bug in `ra/ra_i2c.c`, `readfrom_mem` does a write followed by a read, they both succeed but the bus times out wait on the last TEI. This is just some debug printfs:

```
i2c txi 
i2c txi 
i2c eri 
i2c txi 
i2c txi 
i2c tei 
i2c stopped 
i2c No error 
last stop 0 
i2c txi 
i2c txi 
i2c eri 
i2c rxi 
i2c rxi 
i2c timeout 
```

If you ignore the last TEI you can actually see the register data. Note after this timeout, the state/bus or something locks up (you can repeat this test without a power cycle).

@TakeoTakahashi2020 Might want to take a look at this.

TakeoTakahashi2020, 2024-01-18:
I am sorry for noticing this so late. I will investigate.

sebromero, 2024-01-19:
@TakeoTakahashi2020 Thanks a lot and please let me know if I can help testing something! 🙏

TakeoTakahashi2020, 2024-01-29:
Hi @sebromero san,
Thank you for your issue report.
I have posted a PR #13552 to fix I2C readfrom_mem.
I would appreciate it if you could check with updated file in the PR.

>buses = [I2C(0), SoftI2C(scl=Pin('P408') , sda=Pin('P407'))]

Please note that the same pin cannot be assigned to hardware I2C and soft I2C at the same time.

dpgeorge, 2024-01-31:
Fixed by 81049edf7c67413527ba58a96a3ecfb963addbf9.

closed 2024-01-31 by dpgeorge