micropython/micropython #13285: ampy put command doesn't work

sany3001, 2023-12-28:
I have problem with MicroPython from ESP8266_GENERIC-20231005-v1.21.0.bin. Using ESP-01 (1MB flash) and [programmer](https://www.aliexpress.com/item/32831857167.html?src=google&src=google&albch=shopping&acnt=494-037-6276&slnk=&plac=&mtctp=&albbt=Google_7_shopping&albagn=888888&isSmbAutoCall=false&needSmbHouyi=false&albcp=19374039517&albag=&trgt=&crea=en32831857167&netw=x&device=c&albpg=&albpd=en32831857167&gad_source=1&gclid=CjwKCAiAs6-sBhBmEiwA1Nl8s8HMHah5YaDY71ozKSujrPfCxhAnBRup9_QZMlbfHFohBu1UsM4MvBoCDNIQAvD_BwE&gclsrc=aw.ds&aff_fcid=a8038f7c83a141198a42fc19a8f4f7d9-1703742339402-02513-UneMJZVf&aff_fsk=UneMJZVf&aff_platform=aaf&sk=UneMJZVf&aff_trace_key=a8038f7c83a141198a42fc19a8f4f7d9-1703742339402-02513-UneMJZVf&terminal_id=956eadce909b4f59a9bd428f4f57c7ea&afSmartRedirect=y).
I cannot upload files using

`ampy --port /dev/ttyCH341USB0 put boot.py`
or
`ampy --port /dev/ttyCH341USB0 put boot.py /boot.py`

Always the same exception
`OSError: [Errno 19] ENODEV`

I can use ampy command `run` or `ls` but I cannot use `put`.

Linux Mint environment.
Same error in windows environment.

sany3001, 2023-12-28:
Similar problem when using `pyboard`:

`python3 ./pyboard.py --device /dev/ttyCH341USB0 -f cp boot.py :boot.py`

error:
```
cp boot.py :boot.py
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
```

sany3001, 2023-12-28:
I have switched to adafruit-circuitpython-feather_huzzah-3.1.2.bin (CircuitPython, no more supported for ESP8266) and I was able to upload the file.

robert-hh, 2023-12-28:
Ampy is not maintained any more. Better use mpremote for MicroPython.

sany3001, 2023-12-28:
Same problem as with Thonny. My serial port (CH340) is not visible to mpremote. :-(

sany3001, 2023-12-28:
`mpremote connect /dev/ttyCH341USB0`
works, so fingers crossed now

sany3001, 2023-12-28:
```
mpremote connect /dev/ttyCH341USB0 cp main.py :
cp main.py :
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 19] ENODEV
```

robert-hh, 2023-12-28:
That error code usually pops up if there is not file system or a defective file system on the board.

sany3001, 2023-12-28:
I can run `ls` command well (no error), no files there. I have flashed esp8266 again with MicroPython and no change. So if FS is corrupted, it gets corrupted repeatedly.

It is sad. I have implemented my small project and cannot use it on this old piece of HW.

robert-hh, 2023-12-28:
If the file system is corrupted it stays corrupted until it is rebuilt. The easiest way to force that is erasing the whole flash e.g. with esptool.py and flashing again the firmware. At the first boot MicroPython will recreate the file system. If you use Thonny, it's said that there is a full erase option as well.

sany3001, 2023-12-28:
Thank you for continuous support. I have taken another ESP-01. GPIO0 connected to GND during boot. Erased, flashed MicroPython. Power cycled device. Disabled debug output, trying to copy. Same result.

Have taken 3rd device. Same result.

Blue LED on the board is flashing for a while before error happens. It takes less than 1 second.

I will try now my D1 mini with ESP8266 on board.

sany3001, 2023-12-28:
So all is working well with WeMOS D1 mini (4MB flash) and not working with ESP-01 (1MB flash).
All are same chips (ESP8266).
Because I have 3 ESP-01 I do not believe this is any kind of HW issue.

robert-hh, 2023-12-28:
For the ESP01 with 1M flash you have to use the 1M version of the firmware. In addition, some devices need a different flash mode when flashing the firmware. The default is DIO, but there are three other flash modes as DOUT, QIO, QOUT. Often ESP01 devices need DOUT.

sany3001, 2023-12-28:
I have definitely used wrong BIN file for ESP-01. So this was purely my fault. I also see that there is no FileSystem support in 1MiB version. (I have used DOUT mode.)

What would be really nice is to get some human readable error instead very general exception.
And in case using `ls` command, it would be also good to report that there is no file system rather than returning empty response.

Thank you again for your support! Case closed.

robert-hh, 2023-12-28:
The 1M version of MicroPyhton for ESP8266 has a small file systen  of  392 k size. It's the 512K version that lacks a file system.
```
>>> 
MicroPython v1.22.0 on 2023-12-28; ESP module (1M) with ESP8266
Type "help()" for more information.
>>> import os
>>> os.statvfs("")
(4096, 4096, 98, 95, 95, 0, 0, 0, 0, 255)
```

sany3001, 2023-12-28:
Thanks for addition. I have updated my comment in mean time, removing invalid input.
Just to make it clear - I have tried proper BIN file for 1MiB devices and I was able to upload files (boot.py and main.py) and all is running like a charm!

sany3001, 2023-12-28:
Adding that 1MiB devices do not have FAT FileSystem, 512KiB devices do not have FileSystem at all.

closed 2023-12-28 by sany3001

dpgeorge, 2023-12-29:
ESP8266 1MiB devices use a LittleFS 2, a `os.VfsLfs2`, filesystem.  That makes more efficient use of space than the FAT filesystem (and still works the same with `mpremote`).

> OSError: [Errno 19] ENODEV

ENODEV here means "error, no such device".  That's saying that there is no filesystem available.

sany3001, 2023-12-29:
It makes sense, what you wrote.
However - to have consistent output, it would be better to see error 
> OSError: [Errno 19] ENODEV
also for `ls` command. As this is also file system related operation and there was no output, assuming no error happen.
It can be that this is just normal and its only me who doesn't like it. :-)

robert-hh, 2023-12-29:
If I install the 2MB firmware on a 1MB device, like you did, MicroPython reports a file system with 0 blocks. That causes os,listdir() to return an empty list, and file open operations to fail with `OSError: [Errno 19] ENODEV`. Surely it would be better to let os.listdir() fail as well. But that looks like a low priority issue.
```
MicroPython v1.22.0 on 2023-12-29; ESP module with ESP8266
Type "help()" for more information.
>>> import os
>>> os.statvfs("")
(0, 0, 0, 0, 0, 0, 0, 0, 0, 128)
>>> os.listdir("/")
[]
```