micropython/micropython #13310: rp2: re.match with complicated expression crashes MicroPython requiring hard reset

pdg137, 2023-12-31:
On a Pico with the latest firmware, the follow regex crashes the system completely, requiring a reset / power cycle to recover:

```python
MicroPython v1.22.0 on 2023-12-27; Raspberry Pi Pico with RP2040

Type "help()" for more information.

>>> import re
re.match('[ab]*[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]+[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]+[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]+[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]+[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]*$','cbbbbbbabbabbbbcbbbbbbabbabbbbcbbbbbbabbabbbbcbbbbbbabbabbbbcbbbbbbabbabbb')

```

Of course regular expressions can be arbitrarily complicated, but it seems like there must be a mistake if the board is becoming unresponsive.

On CPython the same command returns a result in a second or two:

```python
Python 3.10.11 (C:\Program Files (x86)\Thonny\python.exe)
>>> import re
re.match('[ab]*[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]+[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]+[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]+[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]+[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]*$','cbbbbbbabbabbbbcbbbbbbabbabbbbcbbbbbbabbabbbbcbbbbbbabbabbbbcbbbbbbabbabbb')
<re.Match object; span=(0, 74), match='cbbbbbbabbabbbbcbbbbbbabbabbbbcbbbbbbabbabbbbcbbb>
>>> 
```

robert-hh, 2024-01-01:
That example succeeds here, but it takes a while.
- on a Teensy 4.x about 50 seconds.
- on a SAMD51 about 7 Minutes.
- on a RP2 about 8:30 Minutes.
```
=== import re
=== re.match('[ab]*[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]+[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]+[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]+[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]+[cb][cb][cb][cb][ab]+[cb][ab]+[cb][ab]+[cb][ab]*$','cbbbbbbabbabbbbcbbbbbbabbabbbbcbbbbbbabbabbbbcbbbbbbabbabbbbcbbbbbbabbabbb')
=== 
=== 
<match num=1>
```

pdg137, 2024-01-01:
Wow, I didn't think of letting it run to completion. So is it just standard that library calls can't be interrupted?

robert-hh, 2024-01-01:
> So is it just standard that library calls can't be interrupted?

It depends. If it's expected that calls can take long, like e.g. uart.read(), they are made interruptible. For re.match() this seems not to be expected.