micropython/micropython #13317: v1.22.0 Extranous empty node in `sys.implementation.version`

Josverl, 2024-01-02:
Since v1.22.0  `sys.implementation.version` has an additional empty node in it
```
>>> import sys;sys.implementation.version
(1, 22, 0, '')
```
I think this is a derelict of the preview naming that should not be there.

jimmo, 2024-01-08:
@Josverl The idea is that this corresponds to the `releaselevel` field that CPython returns here.

i.e.

```
$ python
>>> sys.implementation.version
sys.version_info(major=3, minor=11, micro=6, releaselevel='final', serial=0)
>>> sys.implementation.version[3]
'final'
```

The four options for a release build were:
* Not include the forth field
* Make it `None`
* Make it the empty string
* Some other value (e.g. `"final"`)

We included the field because it makes it easier to write code to check if it's a preview build (without also having to check the field exists). Didn't go with `None` because that seemed strange (and avoided a smaller class of issues like someone assuming it's always a string).

The reason we didn't choose "final" was because "preview" isn't CPython (they have 'alpha', 'beta', 'candidate', or 'final'). It's not clear to me which one of those options corresponds to what we're calling "preview builds".

It's probably not too late to change this though... suggestions welcome!

Josverl, 2024-01-08:
If the intent is to use semver, or to follow version numbers as used on pypi, there should not be a fourth node.
The fourth node must be empty unless there is a pre- or post-release

I find that the code block, to create a human and machine readable version number now needs another adjustment, so I do not agree that the code can be simpler.

I was happy with the return back to semver with 1.19.0 and not dropping the final 0 from that version forward. I think it is a pity to start diverging from the standards again

On a minor note the documentation is not in line with the implementation, but that can be updated

jimmo, 2024-01-09:
> If the intent is to use semver, or to follow version numbers as used on pypi, there should not be a fourth node.

It's a slightly odd situation where MicroPython uses semver, but Python (and Python packages generally) does not.

The only specification we have to follow here is specifically what CPython does: https://docs.python.org/3/library/sys.html#sys.version_info

> The fourth node must be empty unless there is a pre- or post-release

Yes I think this would be true if this tuple was representing a semver.

> 
> I find that the code block, to create a human and machine readable version number now needs another adjustment, so I do not agree that the code can be simpler.

Which code block?

> 
> I was happy with the return back to semver with 1.19.0 and not dropping the final 0 from that version forward. I think it is a pity to start diverging from the standards again

Not quite sure I follow sorry. MicroPython is using semver for all version numbers, e.g. firmware filenames, what's printed on the banner, sys.version, etc. Summarised here: https://github.com/orgs/micropython/discussions/12603

So I don't think we're diverging from any standards (at least not from the semver standard). The intent here was to try and map that onto the scheme that sys.implementation.version uses.

> 
> On a minor note the documentation is not in line with the implementation, but that can be updated

Good point. Will fix.

Josverl, 2024-01-09:
Hi Jim , sorry the code that need to turn sys.implementation.version into the version string used by micropython 
I've not found another **reliable** and cross version way to parse this from other information.
its just another `.rstrip(".")` 

```py
import sys

def version_str(version): # -> str:
    # version >= 1.22.0 have a 4th node in the version tuple that may be ''
    v_str = ".".join([str(n) for n in version]).rstrip(".")
    if (
        v_str
        and v_str.endswith(".0")
        and v_str >= "1.10.0" 
        and v_str <= "1.19.9"
    ):
        # versions from 1.10.0 to 1.20.0 do not have a micro .0
        v_str = v_str[:-2]
    return v_str

print (version_str(sys.implementation.version))
``` 
if there is a better way , or if its just n=1 , then its just a docbug, and easily fixed

dpgeorge, 2024-01-10:
You could try this to get the semver version string:
```py
v_str = ".".join(str(n) for n in version[:3])
if len(version) > 3 and version[3]:
    v_str += "-" + version[3]
```

Josverl, 2024-01-11:
added a PR to bring the documentation up to date.

closed 2024-01-11 by Josverl