micropython/micropython #13335: esp32 c3 pins for ext1 wakeup are not set up correctly

m-cas, 2024-01-03:
in modesp32.h a macro (RTC_VALID_EXT_PINS) defines which pins can be used to wake up the cpu.

The definition is correct for the S2 and S3 variants but it is not for the C3 version that, AFAIK, has the same pins exposed as the S2 and S3 ones.

I am kindly asking for a fix.

Many thanks

ricksorensen, 2024-01-03:
See pull request #13333 from yesterday.   This updates #9583 which has an extensive discussion on topic.

m-cas, 2024-01-03:
@ricksorensen Thanks Rick, alas, I cannot compile my own micropython version, so pls let me know when a binary is available (nightly drop I guess?) and I will immediately test on my xiao esp32c3 board...

jonnor, 2025-03-03:
Should have been fixed in https://github.com/micropython/micropython/pull/15498