micropython/micropython #13342: CPython incompatible: should error on integer literals like "01"

jepler, 2024-01-04:
Recently I noticed that micropython accepts literals like `01` in a source file, while standard Python rejects them saying "SyntaxError: leading zeros in decimal integer literals are not permitted; use an 0o prefix for octal integers".

However, `int("01")` should succeed, parsing the number as decimal.

I implemented this check in a CircuitPython pull request (https://github.com/adafruit/circuitpython/pull/8783) but it feels a bit clunky so I did not pull request the same change in micropython, but filed this issue instead.

andrewleech, 2024-01-04:
It appears that cpython only has this check/error as a means of deprecating the python 2 feature where numbers with leading zeros being interpreted as octal.

> I personally don't think it's worth adding code bulk to micropython just to check / disallow this kind of functionality, though I guess the point is to avoid unintended behaviour when directly running python 2 code.

sosi-deadeye, 2024-01-04:
The error messages are perfect:
```
$ python a.py 
  File "/home/deadeye/a.py", line 2
    b = 01
        ^
SyntaxError: leading zeros in decimal integer literals are not permitted; use an 0o prefix for octal integers
```

`int` must accept strings with leading zeros, except if the base is 0, which lets `int` interpret the input as `int` literal.

```
Traceback (most recent call last):
  File "/home/deadeye/a.py", line 1, in <module>
    a = int("01", 0)
        ^^^^^^^^^^^^
ValueError: invalid literal for int() with base 0: '01'
```

jonnor, 2024-09-01:
Is there a linter that one could with MicroPython code that would catch these? Then we could recommend using that instead of having to implement it at runtime?

andrewleech, 2024-09-01:
> Is there a linter that one could with MicroPython code that would catch these?

`ruff` is configured and used with pre-commit on this repo, it would be sensible to check if it can catch these issues with its existing settings.