micropython/micropython #13362: rmdir should not be able to remove cwd

ubidefeo, 2024-01-05:
I'm working on my wrappers for file tree listing/deleting and found out that if I `os.chdir()` into an empty folder and issue 

```python
os.rmdir('.')
```

the operation is carried out with success, `os.getcwd()` is still valid but `os.listdir()` returns an error because the folder does not exist.

**Steps to reproduce:**
In a **REPL**
```python
os.getcwd()
```

yields 
`'/'`

The following
```python
os.mkdir('test')
os.chdir('test')
os.getcwd()
```

results in
`'/test'`

now
```python
os.rmdir('.')
os.getcwd()
```
still yields
`'/test'`

Of course trying
```python
os.listdir()
```
will result in 
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 2] ENOENT
```
since the folder `/test` does not exist, as a `os.listdir('/')` wil tell us
`[ 'boot.py', 'defaultdict.py', 'file.csv', 'lib', 'mrequests.py', 'urlencode.py', 'urlparseqs.py', 'urlunquote.py']`

fortunately it is still possible to go one up using `os.chdir('..')` or `os.chdir('/')`, but it should not be possible to `os.rmdir()` a folder which is the current `cwd`, same as it's not possible to `rmdir()` a folder which is not empty.

the expected behaviour would be an Error such as the following

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: 39
```
Error 39 is for Directory Not Empty, but I'm not sure an error exists to signal what I want to signal.

Hope someone will look into this :)

u.

stinos, 2024-01-05:
What platform(s) does this behavior occur on?

jimmo, 2024-01-08:
Interestingly CPython doesn't allow you to `os.rmdir('.')` but it will happily allow you to delete the cwd (e.g. `os.rmdir('../a')` where `a` is the cwd).