micropython/micropython #13377: v1.22.1: uart.flush() takes way too long

hmaerki, 2024-01-07:
## Observation

Raspberry Pi pico, v1.21.0, uart.flush() takes 11ms (correct)
Raspberry Pi pico, v1.22.1/0, uart.flush() takes 785ms (way too long)

## How to reproduce

```python
import sys
import time
from machine import UART

print(sys.implementation)

bits_per_s = 9600
uart = UART(0, bits_per_s, bits=8, parity=None, stop=1, tx="GPIO0", rx="GPIO1")

text = "Hello World"
start_us = time.ticks_us()
uart.write(text)
print(f"Duration write() {time.ticks_diff(time.ticks_us(), start_us)/1000:0.2f}ms")

uart.flush()
print(f"Duration write()+flush() {time.ticks_diff(time.ticks_us(), start_us)/1000:0.2f}ms")

bits_per_char = 11 # 1(startbit)+8(bits)+1(stopbit)+1(parity)
print(f"Expected time {len(text)*bits_per_char/bits_per_s*1000:0.2f}ms")
```

## Output
```
(name='micropython', version=(1, 21, 0), _machine='Raspberry Pi Pico with RP2040', _mpy=4358)
Duration write() 0.16ms
Duration write()+flush() **11.41ms** (ok)
Expected time 12.60ms
```

```
(name='micropython', version=(1, 22, 0, ''), _machine='Raspberry Pi Pico with RP2040', _mpy=4614)
Duration write() 0.15ms
Duration write()+flush() **785.37ms** (way too long)
Expected time 12.60ms
```

```
(name='micropython', version=(1, 22, 1, ''), _machine='Raspberry Pi Pico with RP2040', _mpy=4614)
Duration write() 0.20ms
Duration write()+flush() **785.42ms** (way too long)
Expected time 12.60ms
```

robert-hh, 2024-01-07:
It was this commit 28529351aebe45c24e982eaecdba2383d05d2dd4 which changed the wait loop to `mp_event_wait_ms((timeout - now) / 1000);`. It should better call `mp_event_handle_nowait()`.

hmaerki, 2024-01-07:
Hi @robert-hh 

You are a hero!
* May you do a Pull Request with your change?
* If you send me your patched firmware, I can test it in my MODBUS application and the scope.

robert-hh, 2024-01-07:
Fixed firmware is here: https://github.com/robert-hh/Shared-Stuff, files `firmware_PICO_flush.uf2` and `firmware_PICO_W_flush.uf2`.

robert-hh, 2024-01-08:
By the way, instead of `uart.flush()` you can as well use `while not uart.txdone():  pass`.

dpgeorge, 2024-01-15:
Fixed by 07472d05db64a0990b4d47cb001d1a40ba78414f

closed 2024-01-15 by dpgeorge

hmaerki, 2024-01-19:
> Fixed firmware is here: https://github.com/robert-hh/Shared-Stuff, files `firmware_PICO_flush.uf2` and `firmware_PICO_W_flush.uf2`.

Thank you @robert-hh / @dpgeorge  - I successfully tested RPI_PICO-20240117-v1.23.0-preview.47.g16c6bc47c.uf2.