micropython/micropython #13380: rp2: 1.22.0, 1.22.1: Lockup on machine.I2S initialization

roman-yepishev, 2024-01-08:
STR:

1. Flash Raspberry Pi Pico W with `RPI_PICO_W-20231227-v1.22.0.uf2` or `RPI_PICO_W-20240105-v1.22.1.uf2`:
    ```
    MicroPython v1.22.1 on 2024-01-05; Raspberry Pi Pico W with RP2040
    ```

2. Connect over USB serial console (e.g. with `mpremote`).
3. Execute the following (no need to actually connect any I2S device):

    ```
    import machine
    i2s = machine.I2S(0, sck=machine.Pin(13), ws=machine.Pin(14),
                      sd=machine.Pin(15), mode=machine.I2S.TX,
                      bits=16, format=machine.I2S.MONO, rate=8000, ibuf=2000)
    ```

Expected results:

* I2S is initialized and the object is returned.

Actual results:

* The board stops transmitting data over USB serial console and does not return from that call (i.e. if I attempt to play audio afterwards as part of `main.py`, there is no output).

Re-flashing `RPI_PICO_W-20231005-v1.21.0.uf2` allows the use of I2S on the same pins without lockup:

```
MicroPython v1.21.0 on 2023-10-06; Raspberry Pi Pico W with RP2040
Type "help()" for more information.
>>> import machine
>>> i2s = machine.I2S(0, sck=machine.Pin(13), ws=machine.Pin(14),
...                   sd=machine.Pin(15), mode=machine.I2S.TX,
...                               bits=16, format=machine.I2S.MONO, rate=8000, ibuf=2000)
>>> help(i2s)
object I2S(id=0,
...
```

dpgeorge, 2024-01-08:
Thanks for the report.  I can confirm the issue.  It was introduced by commit cfc212b10810863b7dee4f9228ce824ceca837c6, this line:
```c
irq_set_exclusive_handler(DMA_IRQ_0, rp2_dma_irq_handler);
```

The I2S driver uses DMA IRQs, but the new `rp2.DMA` class claims the IRQ handlers for itself.

The fix is probably to change `irq_set_exclusive_handler()` to `irq_add_shared_handler()`.

@nickovs what do you think?

nickovs, 2024-01-08:
I think that should be just fine. The IRQ handlers are only used for Python callbacks, not for any driver processing, so the added latency of the shared handler is unlikely to be noticeable.

dpgeorge, 2024-01-08:
We do have tests for I2S, and they pass on rp2.  But those tests use `I2S(1)` which does still work because it uses DMA IRQ1, and that's different to what the `rp2.DMA` class uses (which is IRQ0).

So until this is fixed it's possible to use `I2S(1)`.

dpgeorge, 2024-01-08:
@nickovs this is a bit more complicated to fix than using `irq_add_shared_handler()`, because the management of enabling/disable the IRQs must also be shared with I2S.

Was there a reason you only supported DMA_IRQ_0 in your driver, and not also DMA_IRQ_1?  Eg if you want to use DMA on both cores then each of them should use a separate IRQ.

nickovs, 2024-01-08:
@dpgeorge No, there was no reason, aside from naivety and not having written a new driver from scratch before.

nickovs, 2024-01-21:
Looking into this a bit more, I think that the way to deal with this is for both the DMA and I2S code to ensure that their chosen DMA interrupt is enabled when they init, but to not switch it off when they deinit. They can then install shared handlers to handle the interrupts on the specific channels that they are using.

For the DMA interrupts there are two separate layers of enablement: there are the `DMA_IRQ_0` and `DMA_IRQ_1` flags in the main NVIC mask and then there are separate, per-channel flags in the DMA controller's `INTE0` and `INTE1` registers. Interrupts will only reach the CPU when a given DMA channel completes if the channel is enabled in `INTEn` and `DMA_IRQ_n` is also enabled. This means that as long as drivers properly set the flags in `INTEn`, not clearing the bit in the interrupt mark when de-initialising shouldn't cause any unnecessary noise but should allow both drivers to call `irq_add_shared_handler()` as needed.

dpgeorge, 2024-01-30:
See #13555 for a fix for this issue.

dpgeorge, 2024-01-31:
Fixed by 807c25d05a81feed1dd0cbc5fde5e0d1ef45c967

closed 2024-01-31 by dpgeorge