micropython/micropython #13387: esp8266: cannot have both a native and a viper function, MemoryError reported

rkompass, 2024-01-08:
The following code should reproduce the issue:

```py
@micropython.native
def f_nat(x):
    return x+1

@micropython.viper
def f_vip(x: int) -> int:
    return x+1

print('native:', f_nat(2), 'viper', f_vip(2))
```
reports:
`MemoryError: memory allocation failed, allocating 100 bytes for native code`

instead of
`native: 3 viper 3` on other platforms.

But both functions consume less than 500 bytes each.

dpgeorge, 2024-01-09:
The above code runs OK for me, pasting it at the REPL, using MicroPython v1.23.0-preview.33.g2ed976f14 on 2024-01-09 -- ESP module with ESP8266.

rkompass, 2024-01-09:
Thank you Damien. You are right, that bug is gone now.
I had used a quite recent MP version, but not the latest and was confident this would not make a difference.

I'm sorry for having made trouble.

So I close this issue as solved.

closed 2024-01-09 by rkompass

robert-hh, 2024-01-09:
@rkompass It may be that IRAM1 has not sufficiently space left in your configuration. Then you can put the code into flash. See `esp.set_native_code_location(start, length)`, which you probably already noticed, because you mention that 500 bytes number. Which b.t.w. was a typical number at the time of writing the documentation.

rkompass, 2024-01-09:
Thank you Robert. I was just continuing my work on a piece of software reporting the capabilities of a MP port, including code generators, after noticing that the wrapping of native/viper/xtensa into try/except is possible if that decorated code is imported from an extra file.
I didn't know of the above function, but now this "bug" makes sense. There was not enough IRAM left.

This raises the questions:

- How can I reset/reuse the IRAM if I have, say a native function, use it, then delete it. Will the IRAM be freed again?

- Is there a way to automatically /programmatically determine a suitable start value for `esp.set_native_code_location(start, length)` ? Length should be 4096 as smallest value, I suppose.

- Given a viper/xtensa function may be put into IRAM, then it should be possible to make it accessible as a hard ISR shouldn't it? Did you consider that (perhaps @dpgeorge)? Given this was possible I could write a fast xtensa routine for quadrature encoding to complete our assortment of these functions. There are none on the esp8266 iirc.

robert-hh, 2024-01-09:
> s there a way to automatically /programmatically determine a suitable start value for esp.set_native_code_location(start, length) ? Length should be 4096 as smallest value, I suppose.

That's hard to tell. `esp.flash_user_start() `tells you the start address of the file system. For a 4M device it's 1M. Below of that is the code, which usually does not fill up all of the area. The 2M firmware uses only 648k. So you could subtract length from that user_start_address as the location for the code. The address given to `esp.set_native_code_location()` is a relative address.

> Given a viper/xtensa function may be put into IRAM, then it should be possible to make it accessible as a hard ISR shouldn't it? 

I used a hard IRQ in a project, but the called function did not use viper code. So you have to try.

robert-hh, 2024-01-09:
The size of the firmware is printed by the command esp.check_fw().

rkompass, 2024-01-09:
I just noted that this latest firmware has only 35K free memory.
But there are `esp.freemem()` which displays the missing 10K and `esp.malloc()`.
And if I malloc 4K (it gives me the address, of course) `esp.freemem()` displays only 6K left, which is plausible.

But overall that seems strange.
As it is not documented: Is this a special version to try certain things out? It seems like a bug. Why should one reduce the already scarce memory further?

dpgeorge, 2024-01-10:
> Is this a special version to try certain things out? It seems like a bug. Why should one reduce the already scarce memory further?

The 10K memory reported by `esp.freemem()` is system memory and is needed in some cases by the system to function.  Eg if you allocate a `socket.socket()` then that memory goes down.

It may be possible to use some of that memory for the Python GC heap, and still have a functioning system.  You can try to increase the size in main.c, the line `STATIC char heap[38 * 1024];`.  Increase that to, say, `42 * 1024`.  But note that this memory is not usable for native/viper code, it's not IRAM.