micropython/micropython #13389: umqtt send config info ,but cant recvice response

BlackCattleProgrammer996, 2024-01-09:
Hello, MicroPython engineer! I am an engineer of Chint Group. I am currently developing applications using MicroPython language on a WIFI module, and one of the functions is MQTT Cilent. When the WIFI module is used as an MQTT client (version 3.1.1) to connect to other MQTT servers (not Aliyun MQTT servers), the connection can be successful, and the publish/subscribe function is normal; As the picture shows:
[image]

However, when the Ali Cloud MQTT server is connected, the device prints a Response timeout (that is, no response data is received) as shown in Figure 2:
[image]

According to the MQTT protocol specification, the MQTT server should respond to data regardless of success or failure, rather than without data. From this phenomenon, it is suspected that the MQTT Connect request initiated by the device has not reached the MQTT server, and we hope that your engineer can help to check and analyze it. If the problem is the MQTT inventory used by the device, can you solve the problem by replacing other MicroPython?

For example, using umqtt.robust2: [https://github.com/fizista/micropython-umqtt.robust2/blob/master/src/umqtt/robust2.py](url)

Or use MicroPython-Mqtt: [https://github.com/peterhinch/micropython-mqtt](url)

BlackCattleProgrammer996, 2024-01-09:
[image]

felixdoerre, 2024-01-16:
Hi, can maybe provide a network capture of the connection, to show if/where the micropython device is misbehaving? The log that you provided is consistent with an MQTT server that just closes the connection after username/password has been sent (which is also, what the error message in the line below says), and so far, I believe this is what happened. So it is not a timeout, the server actively closed the connection. 

> From this phenomenon, it is suspected that the MQTT Connect request initiated by the device has not reached the MQTT server.

I think this is really implausible. As the connection happens over a TCP connection, a message that has not reached the server is treated differently from a timeout.

Maybe the server is expecting an ssl-wrapped connection, and you are connecting in plain (or the other way round)? Maybe Username/Password is wrong and this is how the MQTT server that you are connecting to behaves?