micropython/micropython #13427: ESP8266 NodeMCU 12F V3 - Endless Loop (Wifi?)

rschraml, 2024-01-15:
Hello, 

i tried to use the recent generic firmware (1.22.1) for a ESP8266-12F. Firmware flashing works as expected, i tried different flash modes and baudrates as well - no problem. The issue is, that the ESP gets stuck in an endless boot loop. The WIFI led keeps blinking very fast. I can connect via serial, but i have no chance to enter REPL.

Are there any solutions for this issue?

Thank you in advance

rschraml, 2024-01-15:
If anyone else has this issue - try the 1MB Version of the Firmware. Even if the board has a flash size of 4MB it does not work with the standard version of the firmware. The 1MB version works :-)

closed 2024-01-15 by rschraml

robert-hh, 2024-01-15:
Usually it works with the matching firmware. The Generic one works for all boards with flash >= 2MB. But maybe your board has just 1 MB flash or the flash is defective. You can check the reported flash size with:
`esptool.py --port <USB_port> flash_id`

edgexie, 2024-01-17:
> If anyone else has this issue - try the 1MB Version of the Firmware. Even if the board has a flash size of 4MB it does not work with the standard version of the firmware. The 1MB version works :-)

I use command `esptool --port COM11 --baud 115200 write_flash -fm qio -fs 1MB 0 C:\Users\edgex\Downloads\ESP8266_GENERIC-FLASH_1M-20240105-v1.22.1.bin`, But the wifi led blink also fast, and get garbled code in serial port

edgexie, 2024-01-19:
@rschraml 
I kept pushing the button until I plugged it into the usb. then write  ESP8266_GENERIC-20240105-v1.22.1.bin , It's success.
[image]