micropython/micropython #13432: Pull requests are goign stale

gitcnd, 2024-01-15:
There's 343 contributions sitting in the queue.

If you don't want anyone to help - put up a notice so we know not to waste our time.

If you don't have time to check and merge contributions, appoint someone to help you with that.

robert-hh, 2024-01-15:
> There's 343 contributions sitting in the queue.

That's right.  Quite a few of them are in a common area and will eventually go into a a consolidated feature (e.g. quadrature encoder). A large quantity of them are board definitions. The aspect of these is, that often they do not add anything really different. That applies especially for the rp2 and ESP8266/ESP32 boards, where not even a different board pin mapping is needed. Many of them can be used with one of the GENERIC variants. Each board added will increase the time needed to build the daily previews, and will increase the data storage needed for them.
And quite a few of them are either outdated, in the meaning that they cannot be merged or do not build any more. Keeping a PR recent is a constant challenge. Making a PR is not a fire-and-forget thing.
Processing PR is a lot of work, and Damien does it very careful with extremely valuable feedback. But it takes time, and so he seem to focus on bug fixes and follows his mind of developing MicroPython.
I agree to you, that someone else could be appointed to support Damien, even if they cannot give a definitive answer.

projectgus, 2024-01-15:
There are also 5,469 closed PRs. Although I fully agree that it is frustrating when a PR you submit takes a long time to be reviewed and merged.

The record I have for a PR between submitting and merging is five years. That was *not a MicroPython PR*, in fact it was to a corporate supported project. Five years is absurd and pathological, but some amount of patience is key. :)

In general, please remember that maintainers of open source projects don't owe you their time. It is, as @robert-hh says, a constant challenge and quite a bit of work for the PR submitter. Throwing changes over the wall to become someone else's problem isn't going to work.

@gitcnd I'll make a more specific reply to [your PR](https://github.com/micropython/micropython/pull/13253), which I note is your first PR (thank you!)

closed 2024-01-15 by projectgus