micropython/micropython #13433: pointless build rules needlessly wasting contributor time

gitcnd, 2024-01-15:
Some ideas make sense - but most of them are just mean and rude for no reason:-

error: commit de8bf96: * first word of subject must be capitalised.
error: commit de8bf96: * must match: '^[^!]+: [A-Z]+.+ .+\\.$'
(messages ending with a full stop)
error: commit de8bf96: Subject line must be 72 or fewer characters

please remove all the pointless rules which are making it a royal PITA for contributors.

The less work-for-nor-reason you make us do, the less work YOU need to do to make this better for everyone!

robert-hh, 2024-01-15:
Chris, I do not agree to your rant, even if I stumbled a few times myself over this rule for commit messages. You surely agree that having strict rule for code formatting and code style is a must in projects with many contributors and improves quality and readability. I see the rules for the commit message in the same arena. It helps, when looking through the list of commits. It helps a lot when Damien is putting up the documentation for a new release, which includes information of all commits, consistently formatted. This release documentation is a large piece of work, and it supporting it be just that tiny demand for a proper commit message should not be to much for us.

niftyhacking, 2024-01-15:
I like this type of rule set.
Today a rule parser should be easy to code and invoke and also disabled
(much the same way assert(s) can be used}.
If reading code widely,small things mAtter.

stinos, 2024-01-15:
I for one am really happy that when looking up something in the git history I'm greeted with a concise, consistent list where it's easy to find, read and understand things. In other words these rules aren't for no reason, and while it can take that extra minute to make sure everything is ok, that is for the benefit of many others, now and in the future.

andrewleech, 2024-01-15:
I personally try to follow and ideally enforce these commit rules on every project I'm involved in, personally and at work.

Damien has inspired me to be a better software developer by enforcing a high standard of code, with the cleanliness of the git tree being the first view into this codebase.

I show off the micropython project to all junior developers as an example of best practice git management, with these git message rules being a big part of the consistency of code management.

If you don't have a little time to ensure the quality of your code submission is up to standard, don't bother submitting it, you're just creating extra work for others if you do.

dpgeorge, 2024-01-15:
I consider MicroPython both a journey and a destination: the journey is the stepping stones (git commits) and each step is important in its own right, and the destination is the code as a whole at the latest commit.

closed 2024-01-15 by dpgeorge

projectgus, 2024-01-15:
Would add a suggestion on top of the other comments here, which is that if you haven't already then [installing pre-commit](https://github.com/micropython/micropython/blob/master/CODECONVENTIONS.md#automatic-pre-commit-hooks) really helps shorten the feedback loop for a lot of these checks, so the failure is quickly flagged and corrected. I find this a lot less frustrating than bouncing off the CI because I forgot to run `codeformat.py` manually before pushing.

jimmo, 2024-01-15:
+1 to pre-commit, this has been a huge improvement to my workflow. Thanks @projectgus for setting this up.