micropython/micropython #13439: rp2 PIO StateMachine jump table to produce instructions

jrmoserbaltimore, 2024-01-16:
An enhancement to the Pi Pico StateMachine, add a method producing `jmp` instructions referencing labels in the state machine.  This could then be used to direct a jump to various places without decisions in the PIO program.  This is already possible, just semi-difficult:  as far as I know, there is no way to get the address of a label besides manually counting instructions.

If this is already doable, it's not well documented that I could find, so this becomes just a documentation bug.

For example, consider communicating with the General Instruments AY-3-8930.  This has a simple enough interface and, for the most part, requires writing 8 bits to an address/data bus and 2 bits to a control bus indicating whether the bus is idle, read, write, or address.  The system can communicate with this through the TX FIFO by writing words of `[data][operation][address]`, each 8 bits, but this gets complex:

* The program first writes the address, then must evaluate [operation] to determine if this is read or write
* The program must use `out` to move the contents of the ISR into one register, and then use several `set` and `jmp` instructions to decide where in the program to jump

Instead of these 24 bits, one could use the whole 32 bits in the FIFO, making [operation] a `jmp` instruction:

* The program first writes the address, then sets `x` to 1
* On the next bus clock (next loop of the program), a `jmp (x_dec,"handle data")` directs flow away from setting the address, instead to data handling
* The data handling code uses `out (exec,16)` to jump to the appropriate point in the code

This replaces the coded jump table with a single `out` instruction.

To get the `jmp` instruction, the StateMachine object would provide a member function to produce a `jmp` instruction using the particular label as a destination.

As for the example, this is the output section of a write-only PIO driver for the AY-3-8930.  This uses 16-bit input of an address followed by data, and always writes.

```python
# Set address
label("handle_output")
jmp(x_dec, "set_data")
out(pins,8) .side(0b11)
set(x, 0b1)
jmp("handle_clock")

# Set data
label("set_data")
out(pins,8) .side(0b10) [1]
jmp("handle_clock")
```

To expand this to read/write, I'd need to send another flag, copy it into y, and evaluate it against x:

```python
# Set address
label("handle_output")
jmp(x_dec, "getset_data")
out(pins,8) .side(0b11)
set(x, 0b1)
jmp("handle_clock")

label("getset_data")
out(y,8)
set(x,0b0)
jmp(x_not_y, "get_data")

# Set data
label("set_data")
out(pins,8) .side(0b10) [1]
jmp("handle_clock")

# Get data
label("get_data")
out(pins,8) .side(0b01)
in(pins,8)
jmp("handle_clock")
```

Or by injecting a `jmp`:

```python
# Set address
label("handle_output")
jmp(x_dec, "getset_data")
out(pins,8) .side(0b11)
set(x, 0b1)
jmp("handle_clock")

label("getset_data")
out(exec,16)

# Set data
label("set_data")
out(pins,8) .side(0b10) [1]
jmp("handle_clock")

# Get data
label("get_data")
out(pins,8) .side(0b01)
in(pins,8)
jmp("handle_clock")
```

This saves 2 instructions.  For more complex protocols with multiple possible branch destinations, it saves 2 instructions per possible branch destination.

wodr, 2024-05-30:
Have you tried to print(piogram) ? The instructions are just an array of uint16. You can modify them as you like. Have you though about having 2 programs , one for write one for read? For me this often made the piograms smaller.

nludban, 2025-01-11:
Example workaround to get the label values.  Note these likely need to be adjusted if the program is not loaded at offset 0.

```
import rp2

ws2812_labels = { }

@rp2.asm_pio(sideset_init=rp2.PIO.OUT_LOW,
             out_shiftdir=rp2.PIO.SHIFT_LEFT,
             autopull=True,
             pull_thresh=24)
def ws2812(ws2812_labels=ws2812_labels):
    T1 = 2
    T2 = 5
    T3 = 3
    wrap_target()
    label("bitloop")
    emit = out(x, 1)        .side(0)    [T3 - 1]
    jmp(not_x, "do_zero")   .side(1)    [T1 - 1]
    jmp("bitloop")          .side(1)    [T2 - 1]
    label("do_zero")
    nop()                   .side(0)    [T2 - 1]
    wrap()
    ws2812_labels.update(emit.labels)

print(ws2812_labels)
```

Output: `{'bitloop': 0, 'do_zero': 3}`

Since `asm_pio` executes `ws2812` with [sandboxed globals](https://github.com/micropython/micropython/blob/99ac8193ea299e6d836b076149f561e58226ede6/ports/rp2/modules/rp2.py#L257-L268), `ws2812_labels` access is via a default parameter.  This sandboxing would also make the obvious patch when defining a `label` clumsy to use.  Every "real" instruction returns the `PIOASMEmit` which allows `.side` and `.delay`/`[]` to work, just need to save one and get the labels from it.

```
diff --git a/ports/rp2/modules/rp2.py b/ports/rp2/modules/rp2.py
index 9d13bf1b5..aac9de91f 100644
--- a/ports/rp2/modules/rp2.py
+++ b/ports/rp2/modules/rp2.py
@@ -109,6 +109,7 @@ class PIOASMEmit:
             if label in self.labels:
                 raise PIOASMError("duplicate label {}".format(label))
             self.labels[label] = self.num_instr
+        return self.labels[label]
 
     def word(self, instr, label=None):
         self.num_instr += 1
```

nludban, 2025-01-11:
Note `jmp` is "not supported" in [asm_pio_encode](https://github.com/micropython/micropython/blob/495ce91caefda4623b026520e2b8d8709e274843/ports/rp2/modules/rp2.py#L288).