micropython/micropython #13445: Feature-request: 30% extra flash storage space for free...

gitcnd, 2024-01-16:
I'm considering adding the needful to auto-generate a properly-sized partition table, so there's not so much waste after the end of the micropython firmware.  e.g. I've tested this on my esp32cam, and it works great:-

```
# $IDF_PATH/components/partition_table/Kconfig.projbuild.
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 0x150000,
vfs,      data, fat,     0x160000, 0x2A0000,
```

The "before" space was 2097152, and after is 2752512, which amounts to 31% extra.

There are downsides of course (loss of filesystem anytime it's put someplace else), but for people who want that extra room, it's probably worth-it, since most MCU projects never get changed/updated once their working, and 655360 extra bytes to play with is very useful...

Let me know if you think it's worth doing this and contributing it as a PR

dpgeorge, 2024-01-16:
Reducing the size of the firmware partition for esp32 is not a good idea.  This is because this port grows bigger pretty easily and we don't want to keep changing the filesystem size each release (because it requires users wiping the filesystem).

Eg in the previous release the firmware grew by +35348.  The release before that it grew by +112133.  That's due primarily to updating the IDF.

gitcnd, 2024-01-18:
> There are downsides of course (loss of filesystem anytime it's put someplace else)

so I said - but you missed all my points.

You get that this would be an option the user can invoke from the build commandline, right?  Not the default??

Too add some context to this as well: I'm also working on a new kind of firmware flash utility which does away with the need for an OTA partition (runs from the filesystem - it's complicated code because in the pre-flash prep section, it works out how to reconstruct the firmware file for flashing later, without later using any python or IDF functions [properly handling noncontiguous sectors and littlefs metadata skipping].  This too has a 2 small downsides (a failed flash may not be able to auto-revert back to the working one, and you need to clear enough space off the littlefs or fat32sd filesystem to store (probably gzipped) the new firmware binary...)... but for most projects - temporarily removing "big files" before flashing, then restoring them from the internet afterwards is pretty simple.

I'm pretty sure most would agree that being able to OTA re-flash *any* ESP32 in future would be a nice-to-have feature available, even if you did have to make the space available on the FS to do it.  Better than never being able to at all, right?

closed 2024-01-19 by dpgeorge