micropython/micropython #13456: Wrong checks running on PR's, and one of those 40 fails (unrelated to the PR)

NoosaHydro, 2024-01-17:
Everyone else's PRs show 7 checks were run.

Mine shows 40 !

Something to do with linux thread locking fails:  https://github.com/micropython/micropython/actions/runs/7557557914/job/20577152545?pr=13455

which has nothing to do with my serial port bugfix PR https://github.com/micropython/micropython/pull/13455

jimmo, 2024-01-18:
I have re-run the failing "unix coverage" test on that PR. It's marked as passing now.

Most PRs have ~40 checks, this is typical. 7 would be a very narrow change (e.g. to a specific port).

In this particular case, a change to pyboard.py shouldn't actually trigger many checks, but I think the what-directories-run-which-tests rules could probably use some tuning. It's generally not a big deal.

gitcnd, 2024-01-18:
Thanks jimmo - especially for the explanation: I'm still learning how all these hooks works and what the PR flow is, and that's very helpful to know.

closed 2024-01-18 by dpgeorge

NoosaHydro, 2024-01-18:
(fyi - if you didn't work it out - "gitcnd" and "oceanhydro" are both me.) - I made a new fork in my other account to separate my pyboard contribution apart from my ESP32CAM stuff.  I'll try to learn how to do that from my other one account form now on.

robert-hh, 2024-01-19:
Use branches with git. One can make a merge request from a branch into master. That's what most people do. Then you can keep your local master branch a copy of the official master.
master is just a name. The master branch is not different to other branches one creates.