micropython/micropython #13471: STM32: SPI.read(n) fails if n > 1 (regression)

peterhinch, 2024-01-18:
This arose driving an EEPROM using hard SPI on Pyboard D hardware (soft SPI works OK). Baudrate was 5MHz, other args default. Hardware is a Pyboard D on a DIP28 adaptor on a small PCB with a pair of EEPROM chips on the bus. Traces are very short. This setup has worked in the past when developing [this driver](https://github.com/peterhinch/micropython_eeprom/blob/master/eeprom/spi/SPI.md), and in fact still works with a Pyboard D SF2W running
```
MicroPython v1.18-355-g9ab66b50c-dirty on 2022-04-29; PYBD-SF2W with STM32F722IEK
```
With a Pyboard D SF6W running
```
MicroPython v1.22.1 on 2024-01-05; PYBD-SF6W with STM32F767IIK
```
it behaves as follows. A single byte read from an arbitrary address returns the correct contents, but a multi-byte read returns a bytes object of the correct length but containing zeros. To investigate the fault reading was done with a simple `spi.read(n)`. The hardware was examined with an LA: the single byte and multi-byte reads were identical aside from the extra byte and conformed with the device spec. The MISO result for a 2-byte read was as follows:
[image]

The LA shows correct data being transmitted by the EEPROM but `spi.read(2)` returned `b'\x00\x00'`.

peterhinch, 2024-01-19:
I can't claim to fully grasp the code, but [this line](https://github.com/micropython/micropython/blob/16c6bc47cf2b9247418bca8c0d1361167d08594e/ports/stm32/spi.c#L603) does indicate that multi-byte transfers use DMA, while single byte ones do not. Not a smoking gun, perhaps, but maybe a dripping water pistol...

peterhinch, 2024-01-20:
Unfortunately tying the pin high does not demonstrate the failure. I also tried linking `MOSI` and `MISO` and running the following:
```python
from machine import Pin, SoftSPI, SPI
#spi = SoftSPI(baudrate=5_000_000, sck=Pin('Y6'), miso=Pin('Y7'), mosi=Pin('Y8'))
spi = SPI(2, baudrate=5_000_000)
wbuf = bytearray(b'\xaa' * 10)
rbuf = bytearray(100)
mvb = memoryview(rbuf)
spi.write_readinto(wbuf, mvb[1:11])
print(rbuf[:15])
```
This also failed to demonstrate the issue.

I'd welcome any suggestion how I might be making a mistake here, but as far as I can see the LA trace is textbook correct. It has me completely foxed: I can't see why single byte transfers work and multi byte ones fail; or why hard SPI has problems while soft SPI works with transfers of any length. The latter has been extensively tested.

peterhinch, 2024-01-20:
@andrewleech 

> Is it straightforward for you to test on some more versions, get a closer idea of when the issue came in?

I tested pre-built firmware from the website. V1.18 is the most recent version that passes. V1.19.1 (and later) fail. Hence between January and June 2022. Git blame shows some changes to `spi.c` two years ago.

andrewleech, 2024-01-20:
Assuming it could be a dma config issue; the dma configs are different for each SPI peripheral index (eg. `SPI(0)` vs `SPI(1)`).

Is your failing test on a particular SPI index, or multiple ones?

peterhinch, 2024-01-21:
I am using SPI(2) - the only port available on the DIP28 adaptor (and used on my PCB). Testing SPI(1) would require new hardware - which I could do if this test is of sufficient importance. Please advise.

andrewleech, 2024-01-21:
I was asking about the index just to assist in reproducing and/or looking for code changes related to that index, I'll have a quick play with SPI(2) here and see if I can repro.

Strange that wiring other things to SPI2 isn't reproducing it for you... that could make it harder to track down... 
Did you look at the CS line with the LA by any chance? does it behave correctly in 1 vs multibyte ?

andrewleech, 2024-01-21:
Maybe re-run the failing code against the eeprom, ie. reproduce the failure again, then  print out the pins to ensure they're still configured for spi correctly? eg.
```
>>> Pin('Y6')
Pin(Pin.cpu.B13, mode=Pin.ALT, pull=Pin.PULL_DOWN, alt=Pin.AF5_SPI2)
>>> Pin('Y7')
Pin(Pin.cpu.C2, mode=Pin.ALT, pull=Pin.PULL_DOWN, alt=Pin.AF5_SPI2)
>>> Pin('Y8')
Pin(Pin.cpu.C3, mode=Pin.ALT, pull=Pin.PULL_DOWN, alt=Pin.AF5_SPI2)
```
... and check there's no other code elsewhere accidentally enabling any other pins on AF for SPI2 (should certainly be unlikely, just an idea)

peterhinch, 2024-01-22:
The top trace on the LA is `CS\`. All LA signals are identical with single and multi byte transfers except for the extra clock pulses and MISO data as shown.

I checked for the notion that pins are being reconfigured under working and failing firmware builds.
```
Working firmware 1.18
>>> from machine import Pin
>>> Pin('Y6')
Pin(Pin.cpu.B13, mode=Pin.ALT, pull=Pin.PULL_DOWN, af=Pin.AF5_SPI2)
>>> Pin('Y7')
Pin(Pin.cpu.C2, mode=Pin.ALT, pull=Pin.PULL_DOWN, af=Pin.AF5_SPI2)
>>> Pin('Y8')
Pin(Pin.cpu.C3, mode=Pin.ALT, pull=Pin.PULL_DOWN, af=Pin.AF5_SPI2)
>>> 
Current firmware 1.22.1
>>> Pin('Y6')
Pin(Pin.cpu.B13, mode=Pin.ALT, pull=Pin.PULL_DOWN, alt=Pin.AF5_SPI2)
>>> Pin('Y7')
Pin(Pin.cpu.C2, mode=Pin.ALT, pull=Pin.PULL_DOWN, alt=Pin.AF5_SPI2)
>>> Pin('Y8')
Pin(Pin.cpu.C3, mode=Pin.ALT, pull=Pin.PULL_DOWN, alt=Pin.AF5_SPI2)
``` 
To put this into context I was reluctant to raise this issue especially as the loopback test passed. I therefore went to lengths to be sure there wasn't a silly mistake on my part swapping hardware - even assembling a new PCB - and checking signals with a scope (they look fine). The code is a simple test of the [EEPROM library](https://github.com/peterhinch/micropython_eeprom/blob/master/eeprom/spi/SPI.md). This was developed with hard SPI some years ago. It stopped working after a firmware upgrade.

The following observations seem pertinent.

It works fine with no other change if `HARD` is set to `False`:
```python
            if HARD:
                spi = SPI(2, baudrate=5_000_000)
            else:
                spi = SoftSPI(baudrate=5_000_000, sck=Pin("Y6"), miso=Pin("Y7"), mosi=Pin("Y8"))
```
Observations about the firmware:
1. The firmware handles single transfers without using DMA; single transfers always work.
2. There were changes to the DMA code at the time where the firmware starts to fail.

I'm struggling to see how this can be other than a firmware issue - but I'm as baffled as you by the loopback and "tied high" tests.

robert-hh, 2024-01-23:
I have no EEPROM, so I wired up a flash device to SPI2. The result are mixed. Sometimes it reads data from the flash, sometimes not. Reading from the command area like flash ID fails always. Reading the sfdp area returns sometimes useful data, sometimes it reads 0.

SoftSPI works flawless. Loopback works. Changing phase has no effect.

robert-hh, 2024-01-23:
Another observation: The baudrate cannot be changed. it's always 375 kHz.

peterhinch, 2024-01-23:
@robert-hh Thanks for confirming, but I can't confirm your baudrate observation. The above LA trace shows 9 cycles in 2μs which is close to 5MHz.

I am getting more odd results with hard SPI. With the aim of creating a reproducer I wired up a 74LS73 JK with `clk` connected to `sck` and `Q` connected to `MISO`. `clr\` was connected to pin Y4. The aim was to produce the same 0x55 data as in the above LA trace. I tested with the following.
```python
from machine import SPI, Pin
rst = Pin("Y4", Pin.OUT)
spi = SPI(2, baudrate=5_000_000)
spi.write(b'\x06')
spi.write(b'\x02\0\0\0\0')  # Emulate EEPROM setup
rst(0)
rst(1)  # Reset the JK
spi.read(20)  # b'UUUUUUUUUUUUUUUUUUUU' i.e. good (0x55).
```
The writes have no effect (`MOSI` is unconnected) but I wanted to replicate the same SPI commands as in the driver. The test did not reproduce the fault. I unplugged the Pyboard and plugged it into my EEPROM test rig and it failed. This makes no sense to me.

I then created a build from current source, fully expecting the failure to occur. Yet it worked.

I've just about run out of ideas. The totally consistent takeaways are:
1. Firmware <= 1.18 works.
2. Soft SPI works.
3. Single byte reads (and all writes) work.

I am convinced that there is some kind of marginal condition with hard SPI multi-byte reads, but finding a guaranteed repro is proving to be beyond me.

robert-hh, 2024-01-23:
I just noticed that reading from SPI works if spi.write_readinto() is used and the write and read buffers are different objects. If the buffers are the same, it fails. AFAIK, that's what spi.read() does. It supplies the read buffer as write buffer, which seems to work in most of the cases.

peterhinch, 2024-01-24:
@robert-hh Good find.

I replaced instances of multi-byte reads with `spi.write_readinto()` **with different buffers** and the driver now works with hard SPI on STM32 running latest firmware.

robert-hh, 2024-01-24:
Good. But that should only be a temporary patch.

peterhinch, 2024-01-24:
Sure. On my PC only.

What I'd like to know, when the detailed cause is discovered, is why these three attempts at a repro did not exhibit the fault:
1. Tying MISO to a level (as suggested by @andrewleech).
2. Linking MOSI and MISO.
3. Toggling MISO.

It is a remarkably elusive bug.

robert-hh, 2024-01-24:
I understand why that happens:  it seems to read the output data instead of the input data.  And that is by default 0. In test 2 the data which you read is the one you send. It may be interesting to invert the levels between MOSI and MISO. 
Also: if one defines a different value than 0 as data to be written while reading as second argument to spi.read(), it returns that value. 

There was also the question if SPI(1) is affected as well: it is.

robert-hh, 2024-01-27:
I made that test with an inverter between MOSI and MISO, and sent `spi,read(5, 0x55)`. It failed as expected, returning `b'UUUUU'` instead of the inverted value `b'\xaa\xaa\xaa\xaa\xaa'`.  Reading a single byte works fine, as you stated.

peterhinch, 2024-01-27:
Unfortunately my response a couple of days ago was lost - probably finger trouble on my part. I'll try again :)

Here is a test I did with the JK toggling `MISO`:
```python
MPY: soft reboot
MicroPython v1.22.1 on 2024-01-05; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from machine import SPI, Pin
=== rst = Pin("Y4", Pin.OUT)
=== spi = SPI(2, baudrate=5_000_000)
=== spi.write(b'\x02\0\0\0\0')
=== rst(0)  # Reset the JK
=== rst(1)
=== spi.read(20, 0xFF)
=== 
b'UUUUUUUUUUUUUUUUUUUU'
```
It seems that my hardware isn't producing the same results as yours.

My experience of this bug is that its behaviour is not easily reproduced, nor is it deterministic. When testing with the EEPROM I had locations 0..9 holding 0x55. Subsequent locations were 0, except for one byte which was 0xFF. Printing the first 20 locations consistently failed to return 0x55, but occasionally the location containing 0xFF was correctly returned.

The rock-solid consistent observations I can make are now:
1. Firmware <= 1.18 works.
2. Soft SPI works.
3. Single byte reads (and all writes) work.
4.  `spi.write_readinto()` with different buffers works.

robert-hh, 2024-01-27:
Since looking at the changes to the files related to SPI and DMA did not help, I made simple brute force binary search though the ~600 commits between V1.18 and v1.19 and identified a commit which made it fail.:

`c49d5207e py/persistentcode: Remove unicode feature flag from .mpy file.`

At the commit before that one SPI works fine.
`b295b6f1f py/persistentcode: Remove obsolete comment about qstr window size.`

That seems strange. @peterhinch, maybe you could verify that.

peterhinch, 2024-01-28:
Unfortunately I can't replicate this. The "bad" c49d5207e and "good" b295b6f1f build behave identically. Both behave as follows:
1. Testing with EEPROM: fail. Multi-byte reads return zeros.
2. Testing with MISO tied low: `spi.read(20, 0xff)` returns zeros.
3. MISO toggling: `spi.read(20, 0xff)` returns b'UUUU...'.

However tests 2 and 3 always pass here, even with the latest release build. I have never managed to replicate your results.

robert-hh, 2024-01-28:
That's good and bad. Good, because it is not understandable why this commit should fail. Bad, because w/o replication it's hard to progress. I made another test of building the firmware at the actual master branch commit with the `DEBUG=1`. That disables optimization. This version works here in the flash drive test.

Side notes: 
- I have still a SCL discrepancy. Setting SCL to 10MHz results in a 6Mhz SCL signal.
- The MISO toggling test with an external J/K flip-flop may may have a different active edge vs. data fetch relation than the EEPROM.

peterhinch, 2024-01-28:
I wish we could find some common ground. Here is the most basic test:
1. Install release build PYBD_SF6-20240105-v1.22.1.dfu
2. Connect Y7 (MISO for SPI(2) to gnd.
3. Fire up mpremote and paste the follwing:

```python
from machine import SPI
spi = SPI(2, baudrate=5_000_000)
spi.read(3, 0xff)
```
This is the outcome here. I tied Y7 directly to Gnd, ignoring the JK.
```python
Connected to MicroPython at /dev/ttyACM0
Use Ctrl-] or Ctrl-x to exit this shell
MicroPython v1.22.1 on 2024-01-05; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from machine import SPI
=== spi = SPI(2, baudrate=5_000_000)
=== spi.read(3, 0xff)
=== 
b'\x00\x00\x00'
>>> 
```
Do you see b'\xff\xff\xff'?

> The MISO toggling test with an external J/K flip-flop may may have a different active edge vs. data fetch relation than the EEPROM.

The JK is a 74LS73 which I chose because it is negative edge triggered. With J and K high it therefore reproduces the waveform in my original LA image. The fact that it produces different results from the EEPROM has me completely beat :(

robert-hh, 2024-01-28:
Same as your's here:
```
MicroPython v1.22.1 on 2024-01-05; PYBD-SF6W with STM32F767IIK
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== from machine import SPI
=== spi = SPI(2, baudrate=5_000_000)
=== spi.read(3, 0xff)
=== 
b'\xff\xff\xff'
>>> 
```
So we have a common ground of fails.

robert-hh, 2024-01-28:
Another observation. If I recompile only stm32f7xx_hal_dma.c from stm32_lib, then SPI works.

robert-hh, 2024-01-29:
I changed the Makefile, building the DMA driver with the option -O2.
```
# Don't use -Os for this file, because it causes DMA to fail sometimes for a F7 MPU
# if src and dest in memory are identical. May apply to other MCU's as well.
ifeq ($(MCU_SERIES), f7)
$(BUILD)/lib/stm32lib/STM32$(MCU_SERIES_UPPER)xx_HAL_Driver/Src/stm32$(MCU_SERIES)xx_hal_dma.o: COPT += -O2
endif
```
With that change SPI at the PYBD_SF6 works. It may be needed for other fast MCU's as well.

robert-hh, 2024-01-29:
@peterhinch b.t.w.: Yesterday i had set up a toggle circuitry as well, using a 74HC74 instead of a 74HC73. I did not have the latter in my drawers. That one switches on the rising edge of CLK. That setup show the bug as well.

andrewleech, 2024-01-29:
Oh great find! Does the buffer just need to be declared `volatile` instead?

robert-hh, 2024-01-29:
> Does the buffer just need to be declared volatile instead?
I do not know how that would work. The file for which the code optimization of the compiler was changed is the HAL driver in the STM32 lib for DMA. That driver receives the source and destination addresses as arguments at runtime separately for each transfer direction in the transfer handle.
Secondly, I did/could not want to change the library code, unless it's clear what happens.

andrewleech, 2024-01-29:
At yes good point, I always think changing gcc opt settings like this is only hiding underlying problems, but it's far cleaner than needing to change the library code.

robert-hh, 2024-01-29:
@peterhinch b.t.w. I managed to revive my partially broken PYBD_SF2: Same problem.

peterhinch, 2024-01-30:
@robert-hh I have built #13549 and the EEPROM board now passes the full test, which is quite rigorous. Well done in finding this!

robert-hh, 2024-01-30:
Thank you for verification. I'm still wondering what the difference is. But the only way to tell seems to compile with both options and compare the assembly output.

peterhinch, 2024-01-30:
I'll close this as complete.

I'd be interested to hear if you do find the detailed cause - I'm still puzzled over some of the observations I made.

robert-hh, 2024-01-30:
Comparing the generated code files is pointless. They are completely different, with the -O2 disassembly file about 6.5k smaller than the -Os disassembly file and like 2000 differences. Part of the is related to init/exit code of functions not being iterated.

robert-hh, 2024-01-30:
Adding optimization control statements into the file `stm32f7xx_hal_dma.c` I could tell the function that needs the -O2 option to work. It is `HAL_DMA_Start_IT()`. Again quite reasonable.

projectgus, 2024-01-31:
Suggest keeping this open until a fix is merged.

I don't have a Pyboard D, but I have a few questions if anyone feels like indulging me:

1. If you do the same read from Python twice in a row, does the bug trigger twice or only the first time?
2. If you do a longer read (i.e. `spi.read(40, 0xff)` vs `spi.read(3, 0xff)`) does the bug still trigger? The comments above only seem to include it happening on short reads but this might be a coincidence.
3. If you add some kind of delay before and after the calls to two `HAL_DMA_Start_IT()` in `HAL_SPI_TransmitReceive_DMA()` (stm32f7xx_hal_spi.c), does this bug still trigger? For example, `HAL_Delay(1)`.

robert-hh, 2024-01-31:
1. Doing the same read twice in a row triggers the error twice.
2. The bug triggers for all lengths. With longer reads only the first n bytes may be disturbed, where n varies from attempt to attempt. But we have seen that already before. Example:

```
>>> spi.read(30,0x55);
b'UUUUUUUUUUUUUUUUUUUUUUUUUUUUUU'
>>> spi.read(30,0x55);
b'UUUUUUUUUUUUUUUU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
>>> spi.read(40,0x55);
b'UUUUUUUUUUUUUUUUUUUUUUUUUUUUUUUU\x00\x00\x00\x00\x00\x00\x00\x00'
>>> spi.read(40,0x55);
b'UUUUUUUUUUUUUUUU\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x
```

3. Addingsome kind of delay before and after the two calls to HAL_DMA_Start_IT() does not help. The error is still present.

robert-hh, 2024-02-03:
After two days break, I looked again at the topic and repeated the delay test. Adding a short delay before the first call to HAL_DMA_Start_IT() after line 1808 in stm32f7xx_hal_spi.c works now. The same happens if I insert that delay at the begin of HAL_DMA_Start_IT(). Even HAL_Delay(0) seems sufficient, reducing the delay to upmost three function calls, one assignment and one compare.
Strange! But that complies with the observation, that changing the optimization for either stm32f7xx_hal_spi.c or stm32f7xx_dma_spi.c worked.

projectgus, 2024-02-21:
Thanks to all of the solid work reproducing and narrowing down this issue, I think I've found the root cause (see PR linked above).

If anyone who is able to reproduce on their board can please test again this branch, it'd be much appreciated. :grin:

projectgus, 2024-02-21:
There may be no "smoking gun" bit of assembly for this root cause, due to speculative reads! [From the Cortex-M7 TRM chapter on speculative accesses](https://developer.arm.com/documentation/ddi0489/f/memory-system/speculative-accesses):

> Speculative cache linefills can be initiated to any Cacheable memory address, and in rare cases, regardless of whether there is any instruction that causes the cache linefill.

This likely explains why changing the optimisation level around related code worked around the issue.

robert-hh, 2024-02-21:
> If anyone who is able to reproduce on their board can please test again this branch, I'd be very appreciative. 😁

Done. The fix works in my test.

peterhinch, 2024-02-21:
This fix works for me too.

PR #13717 (2024-03-08): stm32/dma,spi: Fix invalid SPI reads due to D-Cache coherency.
Adds a new DMA API to protect against cache coherency issues on chips with D-Cache, when working with arbitrary buffers (i.e. supplied by Python code).

The API is added to SPI, fixes #13471. Future work may be needed to call this API from other STM32 drivers that use DMA with arbitrary buffers.

Explanation:

1) It's necessary to invalidate D-Cache after a DMA RX operation completes, in case the CPU read (or speculatively accessed) memory in the DMA region during the operation. This seems to have been the root cause of #13471 (and [... cut]