micropython/micropython #13484: missing `micropython-1.22.1.zip`

chenrui333, 2024-01-20:
👋 it looks like 1.22.1 release might not get completed as `micropython-1.22.1.zip` is missing. Thanks!

dpgeorge, 2024-01-22:
The zip file is just a zip of the repository with all submodules checked out, and is only for convenience of those who can't use `git` to fetch the submodules.

GitHub automatically creates source zip/tgz files which contain the repository, excluding the submodules.

Do you specifically need the zip with the submodules included?

chenrui333, 2024-01-28:
> The zip file is just a zip of the repository with all submodules checked out, and is only for convenience of those who can't use `git` to fetch the submodules.
>
> GitHub automatically creates source [...]

Ask mostly for convenience as it was available for the previous releasees? (I still feel like it is good to also attach the release tarball to snapshot the modules) what do you think?

closed 2024-02-20 by dpgeorge