micropython/micropython #13493: Can no longer do `from mpremote import pyboard`

TheChymera, 2024-01-22:
I recently updated from `=mpremote-0.4.0` to `=mpremote-1.22.0`, and apparently this submodule is gone. Any idea where it moved to?

```console
[deco]~ ❱ python
Python 3.11.7 (main, Dec 22 2023, 21:49:07) [GCC 13.2.1 20231216] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mpremote
>>> mpremote.__version__
'1.22.0'
>>> mpremote.pyboard
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'mpremote' has no attribute 'pyboard'
>>> from mpremote import pyboard
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'pyboard' from 'mpremote' (/usr/lib/python3.11/site-packages/mpremote/__init__.py)
>>>
```

I looked through the code and I can't find it. It's mentioned a bunch, but there doesn't seem to be any such submodule anymore.

dpgeorge, 2024-01-22:
What exactly do you need from `pyboard`?  You can probably use the new `mpremote.transport_serial.SerialTransport` class.

TheChymera, 2024-01-22:
@dpgeorge I need

```
    pyb_debug = pyboard.Pyboard(rt_devicenode, 115200)
    pyb_debug.enter_raw_repl()
    result = pyb_debug.exec_raw_no_follow(command)
```

Is that in `SerialTransport` or `PyboardCommand` maybe?

jimmo, 2024-01-29:
I don't recommend using `mpremote.transport_serial.SerialTransport` directly as this is mid-refactoring and it will likely break your application. There is no "official" API (yet!).

@TheChymera If you want to continue to use `pyboard` then you should pin your mpremote dependency at the previous version.

closed 2024-01-29 by TheChymera