micropython/micropython #13495: deque bug ?

pidou46, 2024-01-22:
from doc:
```
class collections.deque(iterable, maxlen[, flags])[¶](https://docs.micropython.org/en/latest/library/collections.html#collections.deque)

    Deques (double-ended queues) are a list-like container that support O(1) appends and pops from either side of the deque. New deques are created using the following arguments:

            iterable must be the empty tuple, and the new deque is created empty.

            maxlen must be specified and the deque will be bounded to this maximum length. Once the deque is full, any new items added will discard items from the opposite end.

            The optional flags can be 1 to check for overflow when adding items.
```
from MP prompt:

```
MPY: soft reboot
MicroPython v1.22.1 on 2024-01-05; Generic ESP32 module with ESP32
Type "help()" for more information.
>>> from collections import deque
>>> b=deque([],100)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: 
>>> 
```

from CPython prompt:
```
Python 3.10.11 (C:\Program Files (x86)\Thonny\python.exe)
>>> from collections import deque
>>> b=deque([],100)
>>> 
```

dpgeorge, 2024-01-22:
It's not a bug, but rather a limitation that you must pass an empty tuple as the first argument.  The docs point this out:

    iterable must be the empty tuple, and the new deque is created empty

dpgeorge, 2024-01-23:
See related #10724.

closed 2024-01-23 by dpgeorge