micropython/micropython #13504: On OSX, mpremote.transport.TransportError: could not enter raw repl

dcmcshan, 2024-01-24:
`% mpremote ls
b'MicroPython v1.20.0-745-g30ba5e11e-dirty on 2024-01-23; ESP32 module (spiram) with ESP32\r\nType "help()" for more information.\r\n>>> '
Traceback (most recent call last):
  File "/Users/danielmcshan/Library/Python/3.9/bin/mpremote", line 8, in <module>
    sys.exit(main())
  File "/Users/danielmcshan/Library/Python/3.9/lib/python/site-packages/mpremote/main.py", line 528, in main
    handler_func(state, args)
  File "/Users/danielmcshan/Library/Python/3.9/lib/python/site-packages/mpremote/commands.py", line 110, in do_filesystem
    state.ensure_raw_repl()
  File "/Users/danielmcshan/Library/Python/3.9/lib/python/site-packages/mpremote/main.py", line 470, in ensure_raw_repl
    self.transport.enter_raw_repl(soft_reset=soft_reset)
  File "/Users/danielmcshan/Library/Python/3.9/lib/python/site-packages/mpremote/transport_serial.py", line 155, in enter_raw_repl
    raise TransportError("could not enter raw repl")
mpremote.transport.TransportError: could not enter raw repl`

but mpremote connects fine:

`% mpremote
Connected to MicroPython at /dev/cu.usbserial-210
Use Ctrl-] or Ctrl-x to exit this shell
MicroPython v1.20.0-745-g30ba5e11e-dirty on 2024-01-23; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> 
`

% sw_vers 
ProductName:		macOS
ProductVersion:		13.0
BuildVersion:		22A380

% python3 --version
Python 3.9.6

conradogarciaberrotaran, 2025-01-20:
This issue is fixed adding a time.sleep(2) in enter_raw_repl

It would look like this:
```python
    def enter_raw_repl(self, soft_reset=True):
        self.serial.write(b"\r\x03")  # ctrl-C: interrupt any running program

        # flush input (without relying on serial.flushInput())
        n = self.serial.inWaiting()
        while n > 0:
            self.serial.read(n)
            n = self.serial.inWaiting()
        time.sleep(2) <<<

        self.serial.write(b"\r\x01")  # ctrl-A: enter raw REPL

```

Same issue on pyboard.py, since the code is exactly the same.

brianhouse, 2025-01-21:
Same issue; the above patch had no effect. 

ProductName:		macOS
ProductVersion:		14.5
BuildVersion:		23F79

Python 3.12.7

mpremote 1.24.1

jabelone, 2025-04-23:
Can you try using a different USB serial adapter or board? I had similar issues and replacing the USB serial adapter fixed it for me (not sure if it was a dodgy driver or hardware).

msbc, 2025-06-03:
I just came here to say that the `time.sleep(2)` "patch" worked for me.

andrewleech, 2026-02-22:
@pavelrevak when you say the issue is not there (on your tool) do you use the same raw mode python code handling as mpremote but with the `time.sleep(2)` patch applied? 
Or have you done something new / different that could inform improvements in mpremote?

New tools / ideas are great to explore options and different ways of working but can also fragment a community if not balanced carefully - the official tool should absolutely be fixed and if the issue can be easlily reproduced this would help to prove the correct fix.

pavelrevak, 2026-02-23:
@andrewleech the main different on MacOS is probably that it switching RTS signals and it resets ESP32. son after boot it not responding for while.

it is really hard to debug mpremote, it has no debug modes, but ook at this:

```
$ ~/.py314/sandbox/bin/mpremote ls          
b'MicroPython v1.27.0 on 2025-12-09; Generic ESP32 module with ESP32\r\nType "help()" for more information.\r\n>>> '
Traceback (most recent call last):
...
mpremote.transport.TransportError: could not enter raw repl
```

there is message after restart and repl prompt, so try this:

```
$ mpremote connect /dev/cu.usbserial-0001 + sleep 2 + ls 
ls :
```

I don't have any large sleep in mpytool but it activelly trying exit app/raw repl and make 200ms wait for response. total time when i trying to wait is about 3seconds. but it starts much sooner about 1s +/-