micropython/micropython #13512: HW Flash Storage

ce4aser, 2024-01-25:
Arduino RP2040 Connect have 16MB flash. But micropython using only 8MB. CircuitPython almost 16MB it is corretly (16MB - circuitpython). 

Please for fast fix this bug. Thank you.

robert-hh, 2024-01-25:
@iabdalkader Was there a specific reason to set the file space to 8MB, like batches with smaller flash?

iabdalkader, 2024-01-25:
> Was there a specific reason to set the file space to 8MB, like batches with smaller flash? Setting it to 15MB works fine, as it seems.

It's just to make sure there's always enough space for code since the flash is shared with the filesystem. I increase the storage to 12MBs leaving 4MBs for code:
https://github.com/micropython/micropython/pull/13513

robert-hh, 2024-01-25:
The code size is at the moment ~600k, and it does not grow exponentially. So there is sufficient headroom for even a larger file system.

iabdalkader, 2024-01-25:
> The code size is at the moment ~600k, and it does not grow exponentially. So there is sufficient headroom for even a larger file system.

We also use the same board config files to build a different firmware image, which takes at least ~1MB. Switching between firmware images with different flash storage size, will break the filesystem, so they must all use the same size. I guess we can increase it to 14MBs leaving 2MBs for the firmware.

PR #13513 (2024-01-26): rp2/boards/ARDUINO_NANO_RP2040_CONNECT: Increase Flash storage space.
Fixes  #13512
closed 2024-01-26 by dpgeorge