micropython/micropython #13521: memory error in `asyncio_tls_server_client`

iot49, 2024-01-26:
I have modified `tests/multi_net/asyncio_tls_server_client.py` to run multiple tests (i.e. echo's) and noticed that after ~8 iterations `tcp_client` crashes. Explicitly running the gc every ~6 iterations avoids the crash. It's also possible to trap the error, in which case the gc is apparently run automatically and the program will continue to operate.

Run on an esp32s3 with 8MBytes of psram and a recent micropython build (december).

My testing code:

```python
import os
import asyncio
import ssl
import gc

HOSTNAME = "micropython.local"
PORT = 8000
FIX = False

cert = cafile = "/certs/rsa_cert.der"
key = "/certs/rsa_key.der"
    
async def handle_connection(reader, writer):
    data = await reader.read(100)
    print("handle_connection echo:", data)
    writer.write(data)
    await writer.drain()
    writer.close()
    await writer.wait_closed()

async def tcp_server():
    server_ctx = ssl.SSLContext(ssl.PROTOCOL_TLS_SERVER)
    server_ctx.load_cert_chain(cert, key)
    server = await asyncio.start_server(handle_connection, "0.0.0.0", PORT, ssl=server_ctx)

async def tcp_client():
    # let server start first
    await asyncio.sleep_ms(100)
    client_ctx = ssl.SSLContext(ssl.PROTOCOL_TLS_CLIENT)
    client_ctx.verify_mode = ssl.CERT_REQUIRED
    client_ctx.load_verify_locations(cafile=cafile)
    for i in range(20):
        reader, writer = await asyncio.open_connection(IP, PORT, ssl=client_ctx, server_hostname=HOSTNAME)
        message = f"msg #{i:3}".encode()
        # required (after every ~ 8th turn)
        if FIX and i % 6 == 0:
            gc.collect()
            print(f"------------------ collect mem_alloc={gc.mem_alloc()}")
        writer.write(message)
        await writer.drain()
        data = await reader.read(100)
        assert data == message
        print("tcp_client read:", data)

async def main():
    await asyncio.gather(tcp_server(), tcp_client())
    print("main DONE")
    
asyncio.run(main())
```

Carglglz, 2024-01-26:
Is this a `MemoryError` or `OSError: ENOMEM` ? 
Also see https://github.com/orgs/micropython/discussions/12316

iot49, 2024-01-26:
no, here's a trace of output:

```
...
tcp_client read: b'msg #  7'
Traceback (most recent call last):
  File "<stdin>", line 68, in <module>
  File "asyncio/core.py", line 1, in run
  File "asyncio/core.py", line 1, in run_until_complete
  File "asyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 65, in main
  File "asyncio/funcs.py", line 1, in gather
  File "asyncio/core.py", line 1, in run_until_complete
  File "<stdin>", line 57, in tcp_client
  File "asyncio/stream.py", line 1, in write
OSError: [Errno 104] ECONNRESET
```

By wrapping individual statement in `try` blocks I identified `writer.write(data)` as the problem, although it could of course also be in the sender or elsewhere. The reason I suspect a memory issue is that running `gc.collect` regularly avoids the problem. 

I've read about the challenges of handling the heap on the esp32. It's one of the reasons I submitted the issue - perhaps it shines light on a lingering problem.

iot49, 2024-01-26:
I was wrong, the error is in `tcp_client`, line `writer.write(message)`.

Carglglz, 2024-01-27:
@iot49 this may be a long shot but could you test #13533? (that `OSError: [Errno 104] ECONNRESET` at client side looks familiar...)

iot49, 2024-01-27:
I'll do this as soon as I can figure out how to compile the VM. 

All but suddenly I am getting a build error:

```
[1279/1309] Building C object esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/usb.c.obj
FAILED: esp-idf/main_esp32s3/CMakeFiles/__idf_main_esp32s3.dir/__/usb.c.obj 
```

This occurs only for `IDF_TARGET=esp32s3` when running

```bash
docker run --rm -v .:/project -w /project espressif/idf:v5.0.4 \
    bash -c "cd micropython/ports/esp32; make clean; make submodules; make V=1 IDF_TARGET=esp32s3"
```

I've asked in the esp32 discussion forum but got no answers. Either something is wrong with my setup (really can't understand what, as I cloned a fresh copy of micropython to investigate) and Espressif's docker always worked fine (same tag). Mysteries!

Otherwise - I have now a PWA (progressive web app) working with an esp32 as the backend server. All communication is encrypted with a Let's Encrypt Certificate. The browser is happy! 

I run the gc before and after every request. I still get random errors, e.g.

```
Traceback (most recent call last):
  File "microdot/microdot.py", line 1289, in handle_request
  File "microdot/microdot.py", line 398, in create
  File "microdot/microdot.py", line 512, in _safe_readline
  File "asyncio/stream.py", line 1, in readline
OSError: -29312
```

but it's mostly working (I did not get these errors before enabling TSL). Huge step forward in terms of security - many thanks to you and all involved for this work!

Carglglz, 2024-01-27:
`OSError: -29312`  that's 
```
include/mbedtls/ssl.h
57:#define MBEDTLS_ERR_SSL_CONN_EOF                          -0x7280
```
That just means that the connection was closed probably the TLS version of `OSError: [Errno 104] ECONNRESET`

iot49, 2024-01-29:
> @iot49 this may be a long shot but could you test #13533? (that `OSError: [Errno 104] ECONNRESET` at client side looks familiar...)

Tried the PR, unfortunately the problem persists.

(Webserver seems now stable, as long as I call the gc very often.)