micropython/micropython #13536: RP2040 I2C bus breaks if pins are changed at runtime

desultory, 2024-01-27:
MicroPython v1.22.1 on 2024-01-05; Raspberry Pi Pico with RP2040

If I attempt to re-init the bus on another set of pins, things break and cannot recover until I do a full hardware reset:

```
>>> from machine import Pin, I2C
>>> a = I2C(1, freq=400000, scl=27, sda=26)
>>> a.readfrom_mem(118, 0xA1, 1)
b'K'
>>> a = I2C(1, freq=400000, scl=3, sda=2)
>>> a.readfrom_mem(118, 0xA1, 1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] EIO
>>> a = I2C(1, freq=400000, scl=27, sda=26)
>>> a.readfrom_mem(118, 0xA1, 1)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] EIO
```

I'm not sure how else I can recover from this other than unplugging it and plugging it back in, soft resets do not correct the issue.

robert-hh, 2024-01-28:
Assigning different Pins to a I2C device does not unassign the previous Pins. That causes trouble and was already discussed previously. You can try to set the previous Pins to Input mode using machine.Pin. That will detach these pins from the I2C device.

robert-hh, 2024-01-28:
At the CC3200 and STM32 ports a method i2c.deinit() is available to some extend. That may be an option here to deliberately release sda and scl. I faintly recall a discussion about that at the old forum, but did not find it. AFAIK,it was considered as not needed.

robert-hh, 2024-01-28:
See also https://github.com/micropython/micropython/issues/6681 for the NRF port.

desultory, 2024-01-28:
> Assigning different Pins to a I2C device does not unassign the previous Pins from the I2C device. That causes trouble and was already discussed previously. You can try to set the previous Pins to Input mode using machine.Pin. That will detach these pins from the I2C device.

This doesn't seem like a great solution.  https://github.com/desultory/bmx280/blob/main/sensor_controller.py this is the code I'm making for reference, I had to remove the ability to loop over pins because of this issue, now it just takes one pin tuple and uses that.

If I were to do that, would I have to track used pins and then turn them off if it doesn't detect devices on that set?

I would like some deinit option for i2c devices that I can just call once I know I'm done with that instance.

robert-hh, 2024-01-29:
> If I were to do that, would I have to track used pins and then turn them off if it doesn't detect devices on that set?

Yes. If you want to change the pin set used by I2C you have to track the pin numbers, because when abaonding the previous pin pair, it has to be set to a state appropriate for your application. An automatic solution can only guess, and the least intrusive state is GPIO input mode. But that may not meet your demand. A i2c.deinit() method could only do the same.
Until then, you can build your own tiny I2C class derived from the built-in machine.I2C class, which keeps track of the pins being used and either offers a deinit method with the Pin states of your choice, or does an automatic deinit if you swap the pins being used. That would be a "Pythonic" solution.

desultory, 2024-02-12:
Is there any reason the rp2 port doesn't have deinit code?

robert-hh, 2024-02-12:
If I would guess: no one missed it so far. It's only implemented for the CC3200 port, which is different in many aspects.