micropython/micropython #13562: Strange bug in try - finally block

AmirHmZz, 2024-01-30:
This snippet is not working properly on Micropython compared to CPython:

```python
class IDGenerator:
    def __init__(self, max_id: int):
        self._i = 0
        self._mi = max_id

    def get(self):
        try:
            return self._i
        finally:
            self._i += 1
            if self._i > self._mi:
                self._i = 0

id_gen = IDGenerator(10)
print(id_gen.get())
```

The output on Micropython `v1.22.1` is:

```python
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 11, in get
AttributeError: 'int' object has no attribute '_i'
```

robert-hh, 2024-01-30:
Confirmed. Somehow, the type of self is changed to int in the line `self._i += 1`, but only after the `finally` statement, and only with the `+=` operator. Writing `self._i = self._i + 1` is fine, and moving `self._i += 1` to another line is fine as well.

dpgeorge, 2024-01-30:
Thanks for the report, I can also confirm the bug.

It's a Python stack overflow in the VM.  The compiler/emitter is not allocating enough Python stack for the try-finally when it contains a return statement.

dpgeorge, 2024-01-31:
Fixed by #13569.

dpgeorge, 2024-01-31:
Fixed by commit 28b18c43fefe6236f899eba79f9a1b2bad134d0f

closed 2024-01-31 by dpgeorge

AmirHmZz, 2024-01-31:
@dpgeorge Thanks for the fix!