micropython/micropython #13566: NimBLE fork has not been merged for two years?

RantyDave, 2024-01-30:
MicroPython v1.22.1 on 2024-01-05; Raspberry Pi Pico W with RP2040

I'm one of the people suffering from random disconnects on Pi Pico bluetooth (https://github.com/micropython/micropython-lib/issues/596) and started picking through the code to see if there's an obvious cause. I note that you build against a fork of NimBLE and that there hasn't been a merge for over two years.

https://github.com/micropython/mynewt-nimble/compare/master...apache%3Amynewt-nimble%3Amaster

Might this help some of the Bluetooth problems?

dpgeorge, 2024-01-30:
The rp2 port (and Pico W board) use BTstack, not NimBLE.

stephanelsmith, 2024-02-01:
Just starting looking into an occasional intermittent ble issue for esp32, is that also the case for esp32 port?

closed 2024-02-22 by RantyDave