micropython/micropython #13600: PWM behavior has changed

ThHupk, 2024-02-05:
PWM frequency now defaults to 0. Used to be 1907. Programs now are obliged to set the frequency.

>>> from machine import PWM
>>> led = PWM('LED')
>>> led.freq()
0

ricksorensen, 2024-02-05:
See now closed pull request #10850.  PWM behavior has been update to be more consistent between ports.  I believe the default behavior is to not start PWM if freq/duty not specified.
https://github.com/micropython/micropython/pull/10850#issuecomment-1528050937

robert-hh, 2024-02-05:
The only port which was not yet aligned is the ESP32 port. There is an open PR #10854 aligning most of the behavior. Only with this port PWM starts at a default frequency of 5000Hz, duty_u16=32768. All other ports now requires freq and duty rate to be set deliberately to get a signal.