micropython/micropython #13605: No support for UART RX IRQ (interrupt)

Omid888, 2024-02-06:
Micropython does not support interrupts on UART, which is very weird since it is a very common protocol and always needed. ([link to the official documentations](https://docs.micropython.org/en/latest/library/machine.UART.html#machine-uart)) 

Alternatives such as asyncIO, PIO and etc are not good enough.
Micropython currently supports this feature only on WiPy boards.

Since I'm using raspberrypi pico, I think their foundation can be a good candidate to apply this request (for the pico boards specifically), but micropython does not support this on almost any boards therefor I'm requesting from the micropython respected team to add this feature.

Thank you for you great work.

jimmo, 2024-02-08:
@Omid888 Which UART interrupt(s) specifically are you interested in? In many cases the "receive data available" (or equivalent) is just used to fill a buffer, which MicroPython's `machine.UART` does for you (configurable with the `rxbuf` kwarg to the constructor).

> Micropython currently supports this feature only on WiPy boards.

The docs need updating, it's actually supported on stm32 and renesas too.

Omid888, 2024-02-12:
1) As I have mentioned I'm using a raspberrypi pico board. Is what you're suggesting applicable to that? Can you provide an example code? (I searched about it in google & gpt4, but no luck!)

2) I need RX UART interrupt handler.

3) Similar to this working code for WiPy:
```
def this_function():
     print("interrupt triggered")

from machine import UART
uart = UART(1, 9600)
uart.irq(UART.RX_ANY, priority=1, handler=this_function, wake=machine.IDLE)

while True:
     pass
 
```

jimmo, 2024-02-13:
@Omid888 

> Is what you're suggesting applicable to that?

Yes. The rp2 port supports the `rxbuf` kwarg to configure buffering on the UART.

> I need RX UART interrupt handler.

Yep, but what for? (I'm not saying we shouldn't support this, but it's good to know why people are using a given feature before adding them)
(In particular, why isn't using `asyncio` or `select`/`poll` applicable?)

Omid888, 2024-02-13:
dear @jimmo
> (In particular, why isn't using asyncio or select/poll applicable?)

Actually because of the fundamental differences between interrupts & asyncIO!

In plain words, the received massage (RX) should be handled right away but it takes time for microController to reach to a place in the codebase that is polling (or receiving data by asyncio) and it's not a good practice to place poll (or asyncio) code snippets in several places of code to achieve that.

andrewleech, 2024-02-13:
asyncio isn't about just using free CPU time around IO use, it's more about conveniently writing and scheduling multiple "mostly concurrent" tasks.

I too was curious as to your actual use case, what your end application is that needs the UART characters handled "immediately" rather than a few ms later by an async task that's written dedicated to responding to your UART input? 

You definitely wouldn't normally put polling or duplicated async handling in multiple places, I generally have individual async tasks for each separate processing need.

Don't get me wrong, asyncio is definitely not "real time", async tasks will happen at varying rates depending on how busy other tasks / background irq's are, so yes there's definitely use cases where immediate attention is needed!

jimmo, 2024-02-15:
Yeah that's the main thing I'm curious about -- the RX IRQ will tell you there's bytes available, not that your specific message is ready (the hardware IRQ trigger has no idea how many bytes you're expecting). But of course your IRQ handler can figure that out.

The neat thing is that doing an `await s.read(23)` will return as soon as there's 23 bytes available, and if the rest of your async program is written in a sufficiently non-blocking way, but of course the benefit to an IRQ is that it pre-emptively runs immediately. We're considering a way to do this with asyncio (e.g. a "high priority" loop running independently of the main loop).

Anyway, UART IRQs is definitely something we can consider adding across more ports.

Omid888, 2024-02-17:
actual use case is this: "A microcontroller connected to an HMI via RS485. The HMI has stop/pause/start/continue buttons on the screen. By hitting any command on the HMI, the uC should immediately execute the command accordingly. The program is -obviously- in a while(True) loop and every iteration could take up to 2 seconds (depending on the conditions)."

Just it.

andrewleech, 2024-02-17:
Interesting, yes responding to rs485 is a useful situation for sure. 

In cases like this, particularly when needing to wait for and detect an assembled packet of data, as Jim suggested earlier being able to await an async read of X number of bytes on an async uart can be a huge benefit, having all the buffering reliably handled for you!

Considering it's only a human facing interface I'd generally consider a response time around 100ms - 200ms as feeling "immediate". I'm not sure if there's an industry standard around this though?

For most of my applications using asyncio tasks will often easily be serviced within 10s of ms, very easily creating immediate feeling user interfaces. I generally only use interrupts when needing to respond to other hardware interfaces etc that need timing accuracy under 10ms.

The main thing that breaks the above async timing estimates is if I'm storing data / logs to flash, especially internal flash, that can introduce much larger latencies.

robert-hh, 2024-02-19:
> Anyway, UART IRQs is definitely something we can consider adding across more ports.

While implementing UART IRQ seems quite straightforward for ports like RP2040, the question is, what should be the trigger selections. I hear the RX data seems wanted, although at high data rates this will create a lot of load. TX done seems to be a good choice as well, called when there is no more data to transmit. Some hardware devices support a RX underrun event, which is raised if the incoming data stream pauses.

robert-hh, 2024-02-21:
I started to implement it for a few ports, beginning with the RP2 port. That works so far, with restrictions caused by the RP2040 hardware design, in that RX and TX interrupts occurs only if the FIFO level crosses a certain boundary, with 4 as the default. As a result of that, receiving less than 4 bytes will not cause an Receive interrupt, and sending less than 4 bytes will not cause an TX interrupt. RX idle interrupts happen even with 1 byte in the FIFO.
In my tests I made another strange observation. The IRQ handler gets the UART object as argument. This Object is almost unusable in the handler, if it is a global. It works if it is a member of a class. Likewise, accessing globals from the IRQ handler fails. That may be related to the problem addressed in PR #11604. Have to test further with other IRQ handlers.

robert-hh, 2024-02-21:
This is the state for RP2:

Trigger names: IRQ_RX, IRQ_RXIDLE, IRQ_TXIDLE
    
- IRQ_RX: The IRQ is triggered when at least 4 bytes are received.
- IRQ_RXIDLE: Works when up to 3 bytes are received and in the FIFO,  but not for longer messages.
- IRQ_TXIDLE: The IRQ is triggered after at least >5 characters are sent at once. It is triggered when the TX FIFO falls below 4 elements.

So it looks not overly useful. The only reliable trigegr would be a combination of IRQ_RX and IRQ__RXIDLE, which will triggger at all incoming data.

robert-hh, 2024-02-23:
About the STM32: For F4xx MCU (e.g. Pyboard), the RXIDLE interrupt triggers twice with every message longer than 1 byte starting with the second message after UART init. The first trigger happens at the first of the message, the second after the message. 1 Byte message get only one trigger, but with different delays after the message. No fix found so far. The only other state I could get was a single trigger for message lengths >= 2, and for 1 byte message alternating trigger/no trigger. maybe it's a known issue with the UART. A RX IRQ can be implemented, That one should work dependable.

 Tests with a F7xx board showed proper behavior.

Omid888, 2024-10-08:
After enough time & struggle, I can say that `buffer` is better than the `interrupt`. Therefore I close the issue.
Thank you all.

P.S: Documentations are still very bad in explaining UART and other modules in micropython. If possible please describe ways of enriching documentations in the repo's readme.md so people can help.