micropython/micropython #13611: Pico-W OSError: [Errno 114] EALREADY

redhead-p, 2024-02-07:
MicroPython v1.22.1 on 2024-01-05; Raspberry Pi Pico W with RP2040

This error occurs as part of a BLE central implementation. I'm using the BLE class direct rather than the aioble library.

Following scanning and a successful _IRQ_PERIPHERAL_CONNECT event, service discovery is initiated.  During service discovery the attribute tree is navigated recording service handles, characteristic handles and descriptor handles. This includes recording handles for the CCC descriptor for characteristics with Notify enabled (UUID = 0x2902).  Following service discovery MTUs are exchanged.  The CCC descriptors then have 1 written to them, to enable notifications to be reported by the server.  The first of these generates an  _IRQ_GATTC_WRITE_DONE with a result of 0 (OK).  The second write causes the OSError 114.

This failure is specific to the Pico-W with the Infineon radio.  The code works as expected without the OSError on both an ESP32 and an Arduino Nano RP2040 Connect (Nina W102 uBlox radio).  I've also tried it with two different peripheral implementations - micropython on the RP2040 Connect and c++/ARM mbed Bluetooth library on an Arduino Nano BLE. The failure happens consistently with both peripheral implementations.

dpgeorge, 2024-02-07:
Are you able to post (ideally a minimal) set of example code for both the central (running on the Pico W) and peripheral?  The idea is to get someone else to reproduce it.

redhead-p, 2024-02-07:
Please find attached code for central on the Pico-W and peripheral.  I've run the peripheral code on an ESP32.  I've tried to minimise the code but as it includes service discovery there's still a lot of it!  I've cut it down to one characteristic with Notify set.  This means that the OSError 114 now occurs on the first read to another characteristic following the single successful write to the CCC descriptor.

[BLE-OSError114.zip](https://github.com/micropython/micropython/files/14196693/BLE-OSError114.zip)

dpgeorge, 2024-02-08:
Thanks for the test code.  I can reproduce the issue and have a fix coming.

dpgeorge, 2024-02-09:
Fixed by b4f59984f7668db457280c1f4007618fd1d235f6

closed 2024-02-09 by dpgeorge

redhead-p, 2024-02-09:
Thanks for the impressively speedy turnaround on this.