micropython/micropython #13634: Pico-W OSError: [Errno 114] EALREADY - Retest

redhead-p, 2024-02-10:
#13611  Retest

MicroPython v1.23.0-preview.114.gb4f59984f on 2024-02-09; Raspberry Pi Pico W with RP2040

I've retested the fix on my live code at it still appears but it appears fixed on the test code as supplied before.  The live code has more than 1 notify enabled characteristic and I've updated the test code for the peripheral to cover this (extra service added at line 108).  The error occurs on the first read as shown below.

[BLE-OSError114-2.zip](https://github.com/micropython/micropython/files/14228853/BLE-OSError114-2.zip)

```
CCCD handle 26 found
CCCD handle 22 found
CCCD handles {26, 22}
cccd handle 26
write result 26 0
cccd handle 22
write result 22 0
Notifications requested
Unhandled exception in IRQ callback handler
Traceback (most recent call last):
  File "<stdin>", line 204, in _ble_event_handler
  File "<stdin>", line 233, in eventHandler
  File "<stdin>", line 416, in _gattc_read_done
  File "<stdin>", line 468, in do_init_reads
OSError: [Errno 114] EALREADY
```

dpgeorge, 2024-02-11:
Thanks for further testing.  This bug is very similar to the previous one, and has a similar fix.  See #13637.

dpgeorge, 2024-02-12:
Should be fixed by abe43fe6873bf78c98634eac74de646f70ec0d32

closed 2024-02-12 by dpgeorge