micropython/micropython #13639: STM32: BLE Bonding Example does not work

hoihu, 2024-02-11:
I'm using a Pyboad SF2 and nightly build as of
`MicroPython v1.23.0-preview.90.ged15b3c6c on 2024-02-06; PYBD-SF2W with STM32F722IEK`

When trying the bonding example:
https://github.com/micropython/micropython/blob/master/examples/bluetooth/ble_bonding_peripheral.py

I got the following response:

```
no secrets available
get secret: 2 0 None
get secret: 10 0 b'irk'
set secret: (10, b'irk') b'\xdc$W\xae\xf1I\xb9\xee\xc0\x9f\xc7\xed,\x82\t\x94'
Traceback (most recent call last):
  File "<stdin>", line 199, in <module>
  File "<stdin>", line 184, in demo
  File "<stdin>", line 81, in __init__
  File "<stdin>", line 158, in _advertise
OSError: -513
```

the problem seems to be related to the ble.config's `addr_mode`. If I uncomment the call here:
https://github.com/micropython/micropython/blob/master/examples/bluetooth/ble_bonding_peripheral.py#L75
and here:
https://github.com/micropython/micropython/blob/master/examples/bluetooth/ble_bonding_peripheral.py#L157
it doesn't throw the error. 

I understand that the ble bonding is still a work in progress demo. However I wanted to have a go on implementing bonding support for the pico_w. , and for this I thought I'd first start with a working example from a supported platform (even if it's from nimble - so another ble stack than on the pico_w).

Also.. is `addr_mode` set to 2 required for ble bonding? 

Thanks for hints and clarifications!

andrewleech, 2024-02-11:
I'm surprised to hear this fails on the pyboard, that's what most official testing would be done on. 
That addr_mode is to generate a resolvable address which I would have expected to be correct: https://docs.openmv.io/library/bluetooth.html#bluetooth.BLE.config

That being said I believe the pyboard-d have a "real" Bluetooth Mac address which with be used by default, instead of needing a generated one. So it should work without this setting.
Indeed the multi-test for the low level Bluetooth pairing (which I think would be run before each release) doesn't appear to have the setting: https://github.com/micropython/micropython/blob/master/tests/multi_bluetooth/ble_gap_pair_bond.py

hoihu, 2024-02-11:
> That being said I believe the pyboard-d have a "real" Bluetooth Mac address which with be used by default, instead of needing a generated one. So it should work without this setting.

Ok, makes sense.

> Indeed the multi-test for the low level Bluetooth pairing (which I think would be run before each release) doesn't appear to have the setting: https://github.com/micropython/micropython/blob/master/tests/multi_bluetooth/ble_gap_pair_bond.py

ah ok, thanks for the pointer to the test.

What I forgot to mention: When removing the ble's `addr_mode` calls, the board properly advertises as `MPY NIMBLE`. When then changing the service to a heart rate monitor (otherwise, it's not visible on the iPhone's bluettooth settings page..) I can connect to it. But I never see `_IRQ_ENCRYPTION_UPDATE` happening (or any of the get_secret / set_secret log messages). Also, when disabling / enabling BLE on the Iphone, it doesn't reconnect automatically either, which should be the case if bonded (?)

Could be that the iPhone blocks the bonding process if the peripheral is not conformal to the service (after all, I had to change it to a heart rate service..). 

My main goal in the end is to read the `current time service` from the iPhone, so my board will autoamtically time-sync when the phone is in proximity. I think I'm really close, since I was even able to scan those services from a pico_w, but I wasn't able to read the charcteristic value, because it needs a bonded connection.

andrewleech, 2024-02-11:
I'm not completely sure (it's been a few years since I worked on this) but whenever you bond the connection the phone will cache the service list and it won't recognise any further changes to services/characteristics until you forget the device on phone and reconnect.

hoihu, 2024-02-11:
Ok so I got the temperature example to work with the following modifications:

- disable the two calls to `self._ble.config(addr_mode=2)` as described.
- on the iPhone, use the "nRF connect" app to connect to the device (since on the dialog via settings->bluettooth, the device will not show up)
- pairing is initiated, `set_secret messages` pop up in the terminal and the device is connected 

up to that part that works. I guess the `addr_mode` not working is probably a minor bug then.

But... I did some further testing afterwards and I observed the following:
- although the `secrets.json` was generated and filled with valid data, it doesn't seem to use them after e.g. a reset
- that means, the bond is lost and I had to manually delete the bonding information on the phone (via bluetooth -> settings-> tap on device and "forget")
- This seems to be another thing though, so maybe I'll open up another issue for that.

hoihu, 2024-02-11:
to conclude on my initial goal of reading out the `current time service` from an Iphone, this code here works ;)
(hastly PoC code..)

```
import sys

sys.path.append("")

from micropython import const

import uasyncio as asyncio
import aioble
import bluetooth

import random
import struct

# org.bluetooth.service.environmental_sensing
_ENV_SENSE_UUID = bluetooth.UUID(0x181A)
# org.bluetooth.characteristic.temperature
_ENV_SENSE_TEMP_UUID = bluetooth.UUID(0x2A6E)
# org.bluetooth.characteristic.gap.appearance.xml
_ADV_APPEARANCE_GENERIC_THERMOMETER = const(768)

# How frequently to send advertising beacons.
_ADV_INTERVAL_MS = 250_000

# Register GATT server.
temp_service = aioble.Service(_ENV_SENSE_UUID)
temp_characteristic = aioble.Characteristic(
    temp_service, _ENV_SENSE_TEMP_UUID, read=True, notify=True
)
aioble.register_services(temp_service)

# Serially wait for connections. Don't advertise while a central is
# connected.
async def peripheral_task():
    while True:
        async with await aioble.advertise(
            _ADV_INTERVAL_MS,
            name="mpy-temp",
            services=[_ENV_SENSE_UUID],
            appearance=_ADV_APPEARANCE_GENERIC_THERMOMETER,
        ) as connection:
            print("Connection from", connection.device)
            try:
                await connection.exchange_mtu(mtu=515)
                print(f'BLE MTU negotiated to be {connection.mtu}')
            except OSError:  # probably [Errno 120] EALREADY
                print(f'BLE MTU already negotiated to be {connection.mtu}')

            await connection.pair()
            cts = await connection.service(bluetooth.UUID(0x1805))
            time_info = await cts.characteristic(bluetooth.UUID(0x2A2B))
            while True:
                time_phone = await time_info.read(timeout_ms=1000)
                if len(time_phone) == 10:
                    year, month, day, hour, minutes, seconds, weekday, frac_sec, reason = struct.unpack('<HBBBBBBBB', time_phone)
                    print(f'{year}-{month}-{day} {hour}:{minutes}:{seconds}')
                          
                await asyncio.sleep_ms(5000)
            await connection.disconnected()
            
# Run tasks.
async def main():
    t2 = asyncio.create_task(peripheral_task())
    await asyncio.gather(t2)

asyncio.run(main())
```

on the console:

```
Connection from Device(ADDR_RANDOM, 76:51:10:76:87:28, CONNECTED)
BLE MTU negotiated to be 185
2024-2-11 16:42:12
2024-2-11 16:42:17
2024-2-11 16:42:22
2024-2-11 16:42:27
2024-2-11 16:42:32
2024-2-11 16:42:37
2024-2-11 16:42:42
```

so to summarize I think there are two issues with the pyboard and BLE:

1) `ble.config(addr_mode)` doesn't seem to work -> this issue. Probably not really important as it seems to work with the normal mac address (as pointed out by @andrewleech )
2) persistent bonding is not working (e.g. after a reset, the bond is lost or the informations in secret.json is not the one that is expceted by the handshake). I'll open another issue for that.

hoihu, 2024-02-11:
Opened https://github.com/micropython/micropython/issues/13642