micropython/micropython #13672: unix port: intermittent infinite loop in import / vfs_reader handling.

andrewleech, 2024-02-15:
We're running unix micropython with `unittest` from micropython-lib to test our application codebase.
This is running micropython @ 4a2e510a

I'm seeing an intermittent problem where the tests sometimes lock up part way through. 

I've finally managed to capture it while running micropython through gdb and the deadlock is during a module import. This happens to be importing an mpy file, not sure if it happens with py files as well at this stage.

Call Stack:
```
mp_reader_vfs_readbyte(void * data) (\code\src\micropython\extmod\vfs_reader.c:55)
read_uint(mp_reader_t * reader) (\code\src\micropython\py\persistentcode.c:154)
load_raw_code(mp_reader_t * reader, mp_module_context_t * context) (\code\src\micropython\py\persistentcode.c:226)
load_raw_code(mp_reader_t * reader, mp_module_context_t * context) (\code\src\micropython\py\persistentcode.c:316)
load_raw_code(mp_reader_t * reader, mp_module_context_t * context) (\code\src\micropython\py\persistentcode.c:316)
mp_raw_code_load(mp_reader_t * reader, mp_compiled_module_t * cm) (\code\src\micropython\py\persistentcode.c:439)
mp_raw_code_load_file(qstr filename, mp_compiled_module_t * context) (\code\src\micropython\py\persistentcode.c:462)
do_load(mp_module_context_t * module_obj, vstr_t * file) (\code\src\micropython\py\builtinimport.c:248)
process_import_at_level(qstr full_mod_name, qstr level_mod_name, mp_obj_t outer_module_obj, _Bool override_main) (\code\src\micropython\py\builtinimport.c:512)
mp_builtin___import___default(size_t n_args, const mp_obj_t * args) (\code\src\micropython\py\builtinimport.c:607)
```

In the current `mp_reader_vfs_readbyte()` function:
```
reader.bufpos = 53
reader.buflen = 0
reader.bufsize = 53
```
so it's returning `MP_READER_EOF (-1)`

One function up, in:
```
STATIC size_t read_uint(mp_reader_t *reader) {
    size_t unum = 0;
    for (;;) {
        byte b = reader->readbyte(reader->data);
        unum = (unum << 7) | (b & 0x7f);
        if ((b & 0x80) == 0) {
            break;
        }
    }
    return unum;
}
```
There's no detection / handling of `0xFF` from `reader->readbyte()` so it's looking forever.

Most times this unit test runs it doesn't show this failure, but every few runs it happens.

I don't know yet where / why it's getting to an EOF on this file when still expecting to read an int, I'll keep digging.

andrewleech, 2024-02-15:
Ok, the readbyte is actually failing above because the file handle has been closed already. I don't yet know _why_ it's being closed when the import reader is still active....

andrewleech, 2024-02-15:
I added a gdb breakpoint in vfs_posix_reader ioctl(MP_STREAM_CLOSE) handling on `close(o->fd);` and set the breakpoint to print a stacktrace and continue (eg. https://blog.mozilla.org/nnethercote/2015/02/13/using-gdb-to-get-stack-traces-at-particular-program-points/) or 
[image]

Once it's locked up, I could see who's calling close on my "file being imported"...
```
Breakpoint 7, vfs_posix_file_ioctl (o_in=0x7ffff7c1c6c0, request=4, arg=0, errcode=0x7fffffff65ac) at ../../extmod/vfs_posix_file.c:193
193	                close(o->fd);
#0  vfs_posix_file_ioctl (o_in=0x7ffff7c1c6c0, request=4, arg=0, errcode=0x7fffffff65ac) at ../../extmod/vfs_posix_file.c:193
#1  0x00005555555a3ae2 in mp_stream_close (stream=<optimized out>) at ../../py/stream.c:433
#2  0x0000555555597000 in fun_builtin_1_call (self_in=0x55555565f1d0 <mp_stream_close_obj>, n_args=<optimized out>, n_kw=<optimized out>, args=0x7fffffff6618) at ../../py/objfun.c:68
#3  0x000055555558e139 in mp_call_function_n_kw (fun_in=fun_in@entry=0x55555565f1d0 <mp_stream_close_obj>, n_args=n_args@entry=1, n_kw=n_kw@entry=0, args=args@entry=0x7fffffff6618) at ../../py/runtime.c:713
#4  0x000055555558e24a in mp_call_function_1 (fun=fun@entry=0x55555565f1d0 <mp_stream_close_obj>, arg=<optimized out>, arg@entry=0x7ffff7c1c6c0) at ../../py/runtime.c:691
#5  0x0000555555590569 in mp_call_function_1_protected (fun=0x55555565f1d0 <mp_stream_close_obj>, arg=0x7ffff7c1c6c0) at ../../py/runtime_utils.c:33
#6  0x0000555555577210 in gc_sweep () at ../../py/gc.c:513
#7  gc_collect_end () at ../../py/gc.c:635
#8  0x0000555555609a55 in gc_collect () at gccollect.c:45

```

andrewleech, 2024-02-15:
vfx_posix_file does indeed have mp_stream_close registered as __del__ and is created with m_new_obj_with_finaliser so it's expected that gc will run the close function which will close the file handle when the vfx_posix_file is gc'd.

However, the vfx_posix_file mp_obj_t should be correctly referenced still as reader->data->file in vfs_reader so I'm not sure why it's getting caught up in a gc sweep... 
```
void mp_reader_new_file(mp_reader_t *reader, qstr filename) {
{
    mp_obj_t file = mp_vfs_open(MP_ARRAY_SIZE(args), &args[0], (mp_map_t *)&mp_const_empty_map);  // this is my vfs_posix_file object
    mp_reader_vfs_t *rf = m_new_obj_var(mp_reader_vfs_t, buf, byte, bufsize);
    rf->file = file;
    reader->data = rf;
}
```
so `rf` is created on the heap, so it'll be seen with the pointer to `file`, however `reader` that's passed in is created on the stack - I'm guessing `gc` can't see this, so it never finds anyone holding a reference to `rf` so probably goes and cleans up both it and `file`.

Even creating `reader` on the heap one call stack up in `mp_raw_code_load_file()` won't help either as it's not tracing to any root pointer.... so `gc` can't see that it's used still....

andrewleech, 2024-02-15:
oh... the `vfs_posix_file_ioctl (o_in=0x7ffff7c1c6c0, request=MP_STREAM_CLOSE, ...)` which closes file handle `0` is being run on a _different_ `o_in` / file object than the current import `(mp_reader_vfs_t *reader)->file` object (also pointing to file handle 0).

So the gc sweep is cleaning up an old / different file object which just so happens to be referencing the same fd number, which is not that surprising as most files are opened on fd 0 and then closed before the next file is opened!

stinos, 2024-02-15:
Sllighty related: https://github.com/micropython/micropython/pull/12652#discussion_r1356769058 i.e. some places compare `fd` to `STDIN_FILENO` which is not necessarily correct, and what you just found also points in the direction that working with fds only can be problematic. Fixing all of properly this would mean switching to using handles instead, at least I don't think there is another way which covers it all.

andrewleech, 2024-02-15:
it keeps getting messier... the `vfs_posix_file_ioctl (o_in=0x7ffff7c1c6c0, request=MP_STREAM_CLOSE, ...)` which closes file handle 0  has a _different_ `o->base.type`. 

The fd 0 that had just been just opened was a `mp_type_vfs_posix_fileio` object, but `vfs_posix_file_ioctl (MP_STREAM_CLOSE)` is being run on a `mp_type_vfs_posix_textio` object (with fd 0)

andrewleech, 2024-02-15:
Found the actual problem I think:
https://github.com/micropython/micropython/blob/809d113dbc37a032e87accc95c140f189733e0c4/extmod/vfs_posix_file.c#L106

`MP_HAL_RETRY_SYSCALL(fd, open(fname, mode_x | mode_rw, 0644), mp_raise_OSError(err)); ` tries to open the file, raises OSError if it fails. On failure, the `mp_obj_vfs_posix_file_t *o = m_new_obj_with_finaliser(mp_obj_vfs_posix_file_t);` made earlier in this function is left hanging to be eventually cleaned up by `gc` which then runs finaliser and closes the (never actually openned) handle 0.

andrewleech, 2024-02-16:
Ok, here's a couple of simple reproductions on the unix port:
`a.py`
```
import gc
import sys
try:
    open("file that doesn't exist", "r")
except OSError:
    pass
gc.collect()  # This will collect the file above and close handle 0

sys.stdin.read(1) # fails with OSError: [Errno 9] EBADF as stdin is handle 0 which has just been closed.
```

`b.py`
```
import gc

try:
    open("file that doesn't exist", "r")
except OSError:
    pass
gc.collect()  # This will collect the file above and close handle 0

# queue up another file object for collecting
try:
    f = open("another file that doesn't exist", "r")
except OSError:
    pass

f = open("/tmp/test", "wb")  # because handle 0 was closed above, this will open on handle 0

gc.collect()  # This will collect the "another" dead file handle above and close f
f.write("test")  # Fails with OSError: [Errno 9] EBADF because the file has been closed
```

andrewleech, 2024-02-16:
And here's a repro of the import deadlock. Files attached: [deadlock.zip](https://github.com/micropython/micropython/files/14303835/deadlock.zip)

`d_src,py` is just a large file to import. The deadlock only happens when importing an mpy.
The issue is triggered by a gc.collect() being implicitely triggered during an import, which will then close the file handle 0. If the file being imported is also fh=0 then it'll be closed part way through the import. 
If this is a .py file it'll generally show as a syntax error as the compiler only sees part of the file. 
If it's an mpy file then reading / running it can crash in a deadlock.

```
mpy-cross -march=x64 d_src.py -o d.mpy
micropython -X heapsize=22K c.py
```
`import d` will be printed, then micropython will be locked up, needing ctrl-c to get out.

dpgeorge, 2024-02-20:
Fixed by 2962e24167a7416d4caf8d25f463e67b46c3a40a

closed 2024-02-20 by dpgeorge