micropython/micropython #13693: tools/mpremote: exec adds extra \ when escape sequences are part of the command string

ubidefeo, 2024-02-17:
I've started digging into an issue I have when I pass multi-line commands to `mpremote exec` and went down my usual rabbit-hole.

## Issue
From `mpremote`'s documentation:
> **exec** - execute the given Python code 

The following command works perfectly inside a REPL session
`exec("import os\nprint(os.listdir())\ntry:\n int('a')\nexcept ValueError:\n print('error')")`

So I assumed that any single string working inside the REPL would work from `mpremote exec`.
I was wrong, because this fails

`mpremote exec "import os\nprint(os.listdir())\ntry:\n int('a')\nexcept ValueError:\n print('error')"`

yielding a 
```
Traceback (most recent call last):
  File "<stdin>", line 1
SyntaxError: invalid syntax
```
Upon suggestion by @iabdalkader , I tried further escaping the `\n` into `\\n`, and simplifying the commands to test if/when it would fail.

This also yields a failure
`mpremote exec "print('hello')\n"`

## Investigation
Believing some form of stripping/sanitising was happening within the `mpremote` code, I pulled the latest changes, made a branch and started messing with the code-base.

Adding a few prints, I found out that the issue is in `exec_raw_no_follow`

```Python
def exec_raw_no_follow(self, command):
    print('exec_raw_no_follow')
    print(f'command: "{command}"')
    if isinstance(command, bytes):
        print('got bytes')
        command_bytes = command
    else:
        print('got string')
        command_bytes = bytes(command, encoding="utf8")
    print(f'command_bytes: {command_bytes}')
```

The output of the above code is the following
```
exec_raw_no_follow
command: "print('ciao')\n"
got string
command_bytes: b"print('ciao')\\n"
```

but can't figure out why `bytes(command, encoding="utf8")` would add an extra `\` to the command received.

If I run the same code in a Python script, no extra `\` is added before `\n`

#### Test code
```Python
command = "print('ciao')\n"
command_bytes = bytes(command, encoding = "utf8")
print(f'command:\n***\n{command}\n***')
print(f'command_bytes: {command_bytes}')
```

#### output
```
command:
***
print('ciao')

***
command_bytes: b"print('ciao')\n"
```

As you can see, the extra `\` is not present in the ouput.

### System and versions
* OS: Mac OS Sonoma 14.2.1
* Python: 3.11.2
* mpremote: 1.22
* micropython: MicroPython v1.23.0-preview.48.g076516d88.dirty

Hope the information above is enough to find what causes the issue, because I can't find an explanation :)

thanks
u.

ubidefeo, 2024-02-18:
@felixdoerre 
thank you for taking the time to explain this.
I'm gonna thoroughly read through it to try and understand it before I formulate an opinion 😃

massimosala, 2024-06-18:
Hi @ubidefeo Ubi

Take attention! If I understood well...

In you test you see no escaping because it is a _str_ object
> command = "print('ciao\n')\n"

Debugging _exec_raw_no_follow_ you see a different result because it is a _bytes_ object:

> command_bytes: b"print('ciao\\n')\\n"

Did I get it right?

jonnor, 2024-08-24:
It seems that the problem has been resolved/clarified. Can we close this issue?

closed 2025-02-04 by projectgus