micropython/micropython #13715: Soft reboot or disabling BLE with active BLE connection hangs ESP32.

xyzzy42, 2024-02-20:
Soft rebooting Micropython while there is an active BLE connection will hang ESP32.  Tested on v1.22.1.

Steps to reproduce:
```python
import bluetooth
ble = bluetooth.BLE()
ble.active(True)
ble.gap_advertise()
# Connect with something, e.g. Nordic nRF Connect app.  Then either:
^D
# or
ble.active(False)
```

This results in a hang requiring a hard reset.   The hang can be avoided by having the connected device disconnect before disabling BLE or the soft reboot.

With NimBLE logging enabled, the connection is closed before hanging:
```
>>> ble.gap_advertise()
I (36653) NimBLE: GAP procedure initiated: advertise;
I (36657) NimBLE: disc_mode=2
I (36661) NimBLE:  adv_channel_map=7 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=800 adv_itvl_max=800
>>> ble.active(False)
I (47005) NimBLE: GAP procedure initiated: stop advertising.
I (47005) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19
[hung]
```

xyzzy42, 2024-02-20:
The hang is here, https://github.com/espressif/esp-nimble/blob/305432c35a9f5e646517a0a895029033fb141b28/porting/nimble/src/nimble_port.c#L262

This code is putting an event on `g_eventq_dflt` and the waiting, forever, for that event to complete.  I think it's trying to wait for any pending events in the queue to empty without a blocking queue drain function.  But this event is never running.  Something probably stopped whatever is supposed to handle this queue.

If I change the forever timeout to a shorter one, then it seems to work ok, since there are probably no events in this queue at shutdown.

dpgeorge, 2024-02-20:
This should have been fixed by c27d304bed737ef0315ca695323f7adc42e61ba9

xyzzy42, 2024-02-20:
Thanks, that was it.  I thought about the GIL and looked on the reboot path for anything that would acquire it.  Didn't realize it was already acquired and the problem was missing code to release it.

dpgeorge, 2024-02-20:
OK, let's close this issue then.

closed 2024-02-20 by dpgeorge