micropython/micropython #13721: asctime gives None in logging

Tico06, 2024-02-21:
### Discussed in https://github.com/orgs/micropython/discussions/13719

<div type='discussions-op-text'>

<sup>Originally posted by **Tico06** February 21, 2024</sup>
Hi There,
I am using logging from standard-lib and trying to format the output with the following:
```
logging.basicConfig(format='%(asctime)s:%(levelname)-8s:%(message)s',
                    level=logging.DEBUG)
```

It results with the following:
```
mpremote rtc --set + mount . + run testing_logging.py
Local directory . is mounted at /remote
None:DEBUG   :My message.
```
How could I get the timestamp heading log line ?
Didn't find any working sample with this.

Thanks and regards,
Eric.</div>

dpgeorge, 2024-02-22:
I think you need to install the `time` module from micropython-lib, eg `mpremote mip install time`.  That will give you access to `time.strftime()` which is needed for the `asctime` logging.

Tico06, 2024-02-22:
Hi George,

Thank you for your message.

This what I did and discussed in the link.

I have created the issue to keep a reminder to, at least, have this documented in the lib, or best, to have it working by its own.

mattytrentini, 2024-02-22:
This is mentioned explicitly in the time README:

https://github.com/micropython/micropython-lib/tree/master/python-stdlib/time#common-uses

But perhaps it should also be added to the logging library?

Tico06, 2024-02-22:
Good point !
A README in logging referencing the one in time would be good.
But still thinking libraries should be working by their own. You may need to install dependencies, that's ok for me, but you shouldn't have to add an import to have it doing the job.

mattytrentini, 2024-02-22:
> But still thinking libraries should be working by their own. You may need to install dependencies, that's ok for me, but you shouldn't have to add an import to have it doing the job.

MicroPython has to concern itself with limited resources; some devices won't have the available memory to install all necessary dependencies for all features. I think if this is documented well it's a reasonable balance.

Tico06, 2024-02-25:
Yes, fully agree with you, a README is enough.
What about a conditional import, if the dependency is installed ?

dpgeorge, 2024-02-26:
> you shouldn't have to add an import to have it doing the job

I'm pretty sure it does not require a special import to make it work.  As long as you have the correct dependency installed (in this case `time`) then the additional logging feature of `asctime` should work.

Tico06, 2024-02-27:
Hi There,
Just tested. After installing the extension lib time, it works as expected.
I'd suggest to keep this ticket open until we have a proper README.
Thx

jonnor, 2024-08-24:
@Tico06  Can you propose a text for adding to the README?

mattytrentini, 2024-08-24:
It's already [been done](https://github.com/micropython/micropython-lib/pull/740). 😄 
I think we can close this.

closed 2024-08-24 by mattytrentini