micropython/micropython #13733: Regression: memoryview() + bytes() fails (__add__)

dhalbert, 2024-02-22:
Tested on PyBoard v1.1.
In v1.21.0:
```py
>>> memoryview(b'x') + b'y'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported types for __add__: 'memoryview', 'bytes'
```
In V1.20.0 and earlier:
```py
>>> memoryview(b'x') + b'y'
array('B', [120, 121])
```
Reversing the args works fine in both:
```py
>>> b'y' + memoryview(b'x')
b'yx'
```
I discovered this by accident when a libraries stopped working in a new version of CircuitPython.

peterhinch, 2024-02-23:
In CPython:
```python
$ python3
Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> memoryview(b'x') + b'y'
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported operand type(s) for +: 'memoryview' and 'bytes'
>>> 
```

dpgeorge, 2024-02-23:
As @peterhinch points out, this was actually intentional, done in commit ca9068e0efc5e53ff7042ef68ad4a4ec9ff91915

It was part of a set of changes to improve inplace and reverse operator support with user types, see also 4b57330465b98df30ef8a10e19a0e197b5797550 and ea7031faff9efe6803d8a8f67ad2e3b4a6d390e3

dhalbert, 2024-02-23:
Well, I did try to test it in CPython (in 3.10.12) even, and thought it was working, but neglected to paste my test. I don't have a record of it in my CPython REPL history, so I don't know what happened. Thanks in any case.

closed 2024-02-23 by dhalbert