micropython/micropython #13797: Not enough HEAP on idf for sending http Requests

ds-digid, 2024-03-01:
### Checks

- [X] I agree to follow the MicroPython [Code of Conduct](https://github.com/micropython/micropython/blob/master/CODEOFCONDUCT.md) to ensure a safe and respectful space for everyone.

- [X] I've searched for [existing issues](https://github.com/micropython/micropython/issues) matching this bug, and didn't find any.

### Port, board and/or hardware

LILYGO_TTGO_LORA32

### MicroPython version

LILYGO_TTGO_LORA32-20240222-v1.22.2.bin

### Reproduction

I have a program that fetches data from a serial interface and send it via http to a server.
As you can see the amount of free heap for the idf is very limitted. When I try to send data via http the memory will drop.

This code is put before calling the http request function and right after the receiption of the answer.

mem = esp32.idf_heap_info(esp32.HEAP_DATA)
print(f"SEND: {mem[5]}")

Output before sending:
SEND: (113840, 38844, 36864, 480)

Output on receive:
RECEIVE: (113840, 4032, 1856, 480)

As 4032 bytes are far to small for the next send process the device crashes.

Is there a way to extend the heap memory for the idf?

### Observed behaviour

2024-03-01T13:17:47 - CRITICAL -  - Task exception wasn't retrieved ; Details: Wifi Internal Error ; IoT-Device version 0.3.2 - Exception:
2024-03-01T13:17:47 - CRITICAL -  - handle_exception - Resetting device
Traceback (most recent call last):
  File "asyncio/core.py", line 1, in run_until_complete
  File "communication/microwifimanager/async_wifi.py", line 158, in _run
  File "communication/microwifimanager/async_wifi.py", line 138, in _connect
OSError: Wifi Internal Error
ets Jun  8 2016 00:22:57

This fails obviosly because of too short memory on the idf heap.

### Additional Information

No, I've provided everything above.

projectgus, 2024-03-05:
Hi @ds-digid,

Recent versions of Micropython will [automatically grow the Python heap as needed](https://github.com/orgs/micropython/discussions/12316), meaning at some point during the request step the ESP32 would otherwise run out of memory for Python so it's expanded the heap (taking from IDF). So the best approach is to figure out which Python code requires a lot of memory and find ways to use less, or less at the one time. If the HTTP response is large then this depend on how it's handled, although not possible to make suggestions without seeing the code.

If there isn't anything obvious, one candidate may be compiling large .py files on demand. You can [freeze packages into the firmware](https://docs.micropython.org/en/latest/reference/packages.html#freezing-packages) or use mpy-cross to produce .mpy files on the host and copy these over instead of the .py files.

ds-digid, 2024-03-05:
Hi @projectgus,
thank you for your response. I checked this out. Python does not run out of heap. There is no problem on this side. I am still precompiling all my files with mpy-cross and work only with .mpy files on the device.
As the same code runs with Micropython version 1.21.0 my guess is, that it is a problem of the underlining idf implementation.

ds-digid, 2024-03-05:
For more information: I use a custom async request library for http requests:

```
import asyncio
import gc
import json

from exceptions import ConnectionError

gc.collect()

# TODO: Remove unneccessary comments for RELEASE

class Response:
    def __init__(self, f, length=None):
        self.raw = f
        self.encoding = "utf-8"
        self._cached = None
        self._length = int(length)

    async def close(self):
        if self.raw:
            self.raw.close()
            await self.raw.wait_closed()
            self.raw = None
        self._cached = None

    @property
    async def content(self):
        if self._cached is None:
            try:
                if self._length is None:
                    self._cached = await self.raw.read()
                else:
                    self._cached = await self.raw.read(self._length)
            finally:
                self.raw.close()
                await self.raw.wait_closed()
                self.raw = None
        return self._cached

    @property
    async def text(self):
        return str(await self.content, self.encoding)

    async def json(self):
        return json.loads(await self.content)

async def request(
    method,
    url,
    data=None,
    loc_json=None,
    headers={},
    stream=None,
    auth=None,
    hostname=None,
    timeout=None,
    parse_headers=True,
):
    redirect = None  # redirection url, None means no redirection
    writer = None
    chunked_data = (
        data and getattr(data, "__iter__", None) and not getattr(data, "__len__", None)
    )

    if auth is not None:
        import ubinascii

        username, password = auth
[... 167 lines cut ...]
async def post(url, **kw):
    try:
        return await request("POST", url, **kw)
    except Exception:
        raise

async def put(url, **kw):
    try:
        return await request("PUT", url, **kw)
    except Exception:
        raise

async def patch(url, **kw):
    try:
        return await request("PATCH", url, **kw)
    except Exception:
        raise

async def delete(url, **kw):
    try:
        return await request("DELETE", url, **kw)
    except Exception:
        raise

```

Carglglz, 2024-03-05:
@ds-digid there is [aiohttp](https://github.com/micropython/micropython-lib/tree/master/python-ecosys/aiohttp) now, which I made to replace this "custom async request" library, so try it and see if the problem persists. 👍🏼

projectgus, 2024-03-05:
> Python does not run out of heap. There is no problem on this side.

Hmm, OK. The point is that instead of running out of heap, Python will claim more memory from IDF - so you don't get any Python MemoryError, but you do get lower free IDF heap instead.

To rule this out as a problem, call `micropython.mem_info()` at the same two times you call idf_heap_info().

ds-digid, 2024-03-06:
@Carglglz: I have tried the aiohttp library but the results are the same. 

Here are some more informations with mem_info() on the specific points. 

- First Block: A complete POST Request

- Second Block: Malfunction with the attempt to reconnect wifi on error.
```
POST: (113840, 39060, 36864, 4)
-------------------------------------
RECEIVE: (113840, 38992, 36864, 4)
stack: 3168 out of 15360
GC: total: 111936, used: 106336, free: 5600, max new split: 36864
 No. of 1-blocks: 1426, 2-blocks: 360, max blk sz: 282, max free sz: 64
+++++++++++++++++++++++++++++++++++++
2024-03-06T07:24:02+01:00 - HEARTBEAT status: 304
stack: 3168 out of 15360
GC: total: 111936, used: 106640, free: 5296, max new split: 36864
 No. of 1-blocks: 1433, 2-blocks: 361, max blk sz: 282, max free sz: 64
```

```
POST: (113840, 39196, 36864, 4)
-------------------------------------
stack: 3680 out of 15360
GC: total: 147904, used: 105984, free: 41920, max new split: 1664
 No. of 1-blocks: 1397, 2-blocks: 349, max blk sz: 282, max free sz: 814
REQUEST: (113840, 3248, 1664, 4)
#######################################
2024-03-06T06:24:29 - ERROR -  - CommMgr - uart_http_requests - No Connection: [Errno 113] ECONNABORTED
Communication failure
Connecting to WiFi
Checking WiFi stability for 3s
Connecting to WiFi
2024-03-06T06:24:43 - CRITICAL -  - Task exception wasn't retrieved ; Details: Wifi Internal Error ; IoT-Device version 0.3.3 - Exception:
stack: 3392 out of 15360
GC: total: 147904, used: 103648, free: 44256, max new split: 1664
 No. of 1-blocks: 1361, 2-blocks: 346, max blk sz: 282, max free sz: 814
2024-03-06T06:24:43 - CRITICAL -  - handle_exception - Resetting device
Traceback (most recent call last):
  File "asyncio/core.py", line 1, in run_until_complete
  File "communication/microwifimanager/async_wifi.py", line 155, in _run
  File "communication/microwifimanager/async_wifi.py", line 129, in _connect
OSError: Wifi Internal Error
ets Jun  8 2016 00:22:57

```

ds-digid, 2024-03-06:
Btw, is there any chance to print out the installed idf version?

robert-hh, 2024-03-06:
```
import platform
print(platform.platform())
```

ds-digid, 2024-03-06:
> ```
> import platform
> print(platform.platform())
> ```

Thank you

projectgus, 2024-03-07:
Thanks @ds-digid . 

From the numbers above, between "before" and "after" snapshots Micropython finds it has run out of heap and grows the total Python heap from 111936 to 147904 to avoid failure (+35968). This drops the available IDF memory from 39196 to 3248, causing the Wi-Fi internal failure.

Do you have an idea of what is using 106640 bytes of Python heap in the "before" snapshot? If you can find the cause of that, before the request even starts, and get this usage number down, then the subsequent failure will go away.

ds-digid, 2024-03-08:
@projectgus 
The 106640 bytes come mainly from the imports at startup. Micropython seems not very efficient there. Some parts take an incredible amount of Heap memory.
For example:
importing the datetime library (previously installed via mip) takes ~20kB of heap  memory. (I need this library because micropython builtin time module lacks compatibility with unix timestamps)
After all imports from all submodules in my system (With very often calls of gc.collect() to reduce memory consumption):

```
$Init System
stack: 2288 out of 15360
GC: total: 127936, used: 82176, free: 45760, max new split: 17408
 No. of 1-blocks: 935, 2-blocks: 280, max blk sz: 282, max free sz: 2740
```

So that there is not that much free heap space for the running program.

projectgus, 2024-03-13:
> The 106640 bytes come mainly from the imports at startup. Micropython seems not very efficient there. Some parts take an incredible amount of Heap memory. For example: importing the datetime library [...]

Right, thanks for the extra info. There's a few things you can do here, depending on how deep you want to go. It seems like your current setup almost has enough RAM, so with luck you don't need to change much to get it all working smoothly:

1. Write some workaround functions instead of pulling in the full datetime module. If all you need is to convert between Unix and Micropython Epoch time stamps, this [can be done by adding or subtracting a constant](https://stackoverflow.com/a/60180717).
2. Make a custom build that [freezes the main dependencies into the firmware](https://docs.micropython.org/en/latest/reference/manifest.html#summary). They will use a lot less RAM compared to loading from the filesystem. Unfortunately this requires a MicroPython build environment. There is [an open PR to add a feature allowing bytecode can be loaded via mpremote without using RAM](https://github.com/micropython/micropython/pull/8381), but not available yet.
3. Rewrite your code to reduce memory usage. One thing that can help is shortening names of variables, functions, etc. Each name is only stored in RAM once (unless frozen), but if you have a lot of unique long names then it can add up.

There are some similar discussions in other threads which may have other tips, [for example](https://github.com/orgs/micropython/discussions/14056#discussioncomment-8752032).

> The imports of this 3 classes alone take ~4kB of RAM. This is just insane in my view. There is not that much going on...

It can be a bit tricky to accurately measure this kind of thing. In particular, the first time you import something from the filesystem then there may be some additional filesystem initialisation that happens in the background and uses some RAM.

I compiled those classes down to a 790 byte .mpy file with mpy-cross (mostly strings for names), and as best as I can tell importing it uses 1760 bytes of RAM on ESP32 (although some of this is probably space for strings that isn't all used, so the next import can use some of that too.)

To accurately measure I also created a 0 byte file empty.py and compiled it to empty.mpy, to take out any initialisation costs from the first import.

<details>
<summary>Full session log - with and without empty.mpy - if you're interested</summary>

```
MPY: soft reboot
MicroPython v1.23.0-preview.34.g5785c6658d.dirty on 2024-01-10; Generic ESP32 module with ESP32
Type "help()" for more information.
>>> import micropython, gc
>>> gc.collect()
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 56000, used: 1152, free: 54848, max new split: 110592
 No. of 1-blocks: 20, 2-blocks: 6, max blk sz: 18, max free sz: 3406
>>> import scmsg
>>> gc.collect()
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 56000, used: 3440, free: 52560, max new split: 110592
 No. of 1-blocks: 41, 2-blocks: 15, max blk sz: 19, max free sz: 3260
>>> 
MPY: soft reboot
MicroPython v1.23.0-preview.34.g5785c6658d.dirty on 2024-01-10; Generic ESP32 module with ESP32
Type "help()" for more information.
>>> import micropython, gc
>>> gc.collect()
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 56000, used: 1152, free: 54848, max new split: 110592
 No. of 1-blocks: 20, 2-blocks: 6, max blk sz: 18, max free sz: 3406
>>> import empty
>>> gc.collect()
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 56000, used: 1856, free: 54144, max new split: 110592
 No. of 1-blocks: 27, 2-blocks: 7, max blk sz: 18, max free sz: 3339
>>> import scmsg
>>> gc.collect()
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 56000, used: 3616, free: 52384, max new split: 110592
 No. of 1-blocks: 47, 2-blocks: 16, max blk sz: 19, max free sz: 3241
```
</details>